ChristophHaag/SteamVR-OpenHMD #1: SteamVR MacOS support?

oscarbg, 2017-11-25:
Hi,
I have Oculus DK2 which is supported by OpenHMD and would like to try to add Metal2 (Macos) support for a simple OpenVR app and running on Macos..
notice I don't even know if OpenHMD compiles/works on Macos but if yes then is your project compatible with SteamVR for Macos (which supports Macos 10.13 High sierra)?

thanks..

ChristophHaag, 2017-11-25:
I don't know much about Mac OS X but in general OpenHMD should work on Mac OS X. Recently I've seen someone complain about OpenHMD not matching hidapi paths correctly or something.

SteamVR-OpenHMD is 100% untested on anything but Linux, and I'm unsure if anyone else has even tried it yet. :)

I do believe it should "just work" cross platform, except for maybe this configuration I hacked together: https://github.com/ChristophHaag/SteamVR-OpenHMD/blob/master/ohmd_config.h

If you don't have controllers and the "config file" thing doesn't work, you probably want to edit it to say

```
else {
        DriverLog("could not open config file %s, using defaults\n", filename);
        vals[0] = 0; // openhmd index for the hmd to use for the display setup
        vals[1] = 0; // openhmd index for the tracker of the hmd, in case with Nolo or so it's a different tracker
        vals[2] = -1; // openhmd index of the left controller, -1 disables it
        vals[3] = -1; // openhmd index of the right controller
```
or something like that.

---

Anyway, since OpenHMD doesn't have positional tracking support for the DK2 yet (soon™), if you don't have Nolo or other external positional trackers there may be alternatives that work with the old 0.5.0.1 Oculus SDK that was still cross platform.

* SteamVR may already come with an oculus legacy driver in Steam/SteamApps/common/SteamVR/drivers/oculus_legacy/. Currently that only has empty directories (?) on my linux system but I imagine that it would work with the old Oculus SDK. I never looked into that one though.
* As far as I know, https://github.com/OSVR/OSVR-Oculus-Rift still supports the old Oculus SDK, so taking a detour over the OSVR SDK server, together with OSVR's SteamVR plugin https://github.com/OSVR/SteamVR-OSVR could give you positional tracking, though OSVR's setup is not super straight forward.

oscarbg, 2017-12-08:
thanks for all information..
I'm gonna try all options and report..

marl0n, 2018-12-06:
Any luck? Could I get a Samsung Odyssey+ working on OSX within FCPX trough OpenHMD?