ChristophHaag/SteamVR-OpenHMD #3: error on make

tekjanson, 2018-05-12:
[  2%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/openhmd.c.o
[  4%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/platform-win32.c.o
[  6%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_dummy/dummy.c.o
[  8%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/omath.c.o
[ 10%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/platform-posix.c.o
[ 12%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/fusion.c.o
[ 14%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/shaders.c.o
[ 16%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_oculus_rift/rift.c.o
[ 18%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_oculus_rift/packet.c.o
[ 20%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_deepoon/deepoon.c.o
[ 22%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_deepoon/packet.c.o
[ 24%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_wmr/wmr.c.o
[ 26%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_wmr/packet.c.o
[ 28%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_psvr/psvr.c.o
[ 30%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_psvr/packet.c.o
[ 32%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_htc_vive/vive.c.o
[ 34%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_htc_vive/packet.c.o
In file included from /home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openhmd/src/drv_htc_vive/packet.c:2:0:
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openhmd/src/drv_htc_vive/vive_config.h:8:9: warning: unknown option after ‘#pragma GCC diagnostic’ kind [-Wpragmas]
 #pragma GCC diagnostic ignored "-Wmisleading-indentation"
         ^
[ 36%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/ext_deps/mjson.c.o
[ 38%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_nolo/nolo.c.o
[ 40%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_nolo/packet.c.o
[ 42%] Building C object subprojects/openhmd/CMakeFiles/openhmd-shared.dir/src/drv_external/external.c.o
[ 44%] Linking C shared library libopenhmd.so
[ 44%] Built target openhmd-shared
[ 46%] Building CXX object CMakeFiles/driver_openhmd.dir/driverlog.cpp.o
In file included from /home/tekjanson/Documents/Personal/SteamVR-OpenHMD/driverlog.h:9:0,
                 from /home/tekjanson/Documents/Personal/SteamVR-OpenHMD/driverlog.cpp:3:
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1538:71: error: ‘nullptr’ was not declared in this scope
   virtual bool Sync( bool bForce = false, EVRSettingsError *peError = nullptr ) = 0;
                                                                       ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1540:118: error: ‘nullptr’ was not declared in this scope
  *pchSection, const char *pchSettingsKey, bool bValue, EVRSettingsError *peError = nullptr ) =
                                                                                    ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1541:122: error: ‘nullptr’ was not declared in this scope
 chSection, const char *pchSettingsKey, int32_t nValue, EVRSettingsError *peError = nullptr ) =
                                                                                    ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1542:121: error: ‘nullptr’ was not declared in this scope
 pchSection, const char *pchSettingsKey, float flValue, EVRSettingsError *peError = nullptr ) =
                                                                                    ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1543:129: error: ‘nullptr’ was not declared in this scope
 ion, const char *pchSettingsKey, const char *pchValue, EVRSettingsError *peError = nullptr ) =
                                                                                    ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1547:105: error: ‘nullptr’ was not declared in this scope
 l( const char *pchSection, const char *pchSettingsKey, EVRSettingsError *peError = nullptr ) =
                                                                                    ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1548:109: error: ‘nullptr’ was not declared in this scope
 2( const char *pchSection, const char *pchSettingsKey, EVRSettingsError *peError = nullptr ) =
                                                                                    ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1549:107: error: ‘nullptr’ was not declared in this scope
 t( const char *pchSection, const char *pchSettingsKey, EVRSettingsError *peError = nullptr ) =
                                                                                    ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1550:160: error: ‘nullptr’ was not declared in this scope
 , VR_OUT_STRING() char *pchValue, uint32_t unValueLen, EVRSettingsError *peError = nullptr ) =
                                                                                    ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1552:83: error: ‘nullptr’ was not declared in this scope
   virtual void RemoveSection( const char *pchSection, EVRSettingsError *peError = nullptr ) = 
                                                                                   ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1553:116: error: ‘nullptr’ was not declared in this scope
 n( const char *pchSection, const char *pchSettingsKey, EVRSettingsError *peError = nullptr ) =
                                                                                    ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:2038:94: error: ‘nullptr’ was not declared in this scope
 oid *GetGenericInterface( const char *pchInterfaceVersion, EVRInitError *peError = nullptr ) =
                                                                                    ^
In file included from /home/tekjanson/Documents/Personal/SteamVR-OpenHMD/driverlog.h:9:0,
                 from /home/tekjanson/Documents/Personal/SteamVR-OpenHMD/driverlog.cpp:3:
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:2187:146: error: ‘nullptr’ was not declared in this scope
  ulContainer, vr::ETrackedDeviceProperty prop, vr::ETrackedPropertyError *peError = nullptr );
[... 29 lines cut ...]
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:2834:34: error: ‘nullptr’ was not declared in this scope
    if ( m_pVRServerDriverHost == nullptr )
                                  ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h: In member function ‘vr::IVRWatchdogHost* vr::COpenVRDriverContext::VRWatchdogHost()’:
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:2844:30: error: ‘nullptr’ was not declared in this scope
    if ( m_pVRWatchdogHost == nullptr )
                              ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h: In member function ‘vr::IVRDriverLog* vr::COpenVRDriverContext::VRDriverLog()’:
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:2854:27: error: ‘nullptr’ was not declared in this scope
    if ( m_pVRDriverLog == nullptr )
                           ^
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h: In member function ‘void vr::COpenVRDriverContext::Clear()’:
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:2931:19: error: ‘nullptr’ was not declared in this scope
   m_pVRSettings = nullptr;
                   ^
[... cut ...]
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:2978:23: error: ‘nullptr’ was not declared in this scope
   VRDriverContext() = nullptr;
                       ^
CMakeFiles/driver_openhmd.dir/build.make:62: recipe for target 'CMakeFiles/driver_openhmd.dir/driverlog.cpp.o' failed
make[2]: *** [CMakeFiles/driver_openhmd.dir/driverlog.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/driver_openhmd.dir/all' failed
make[1]: *** [CMakeFiles/driver_openhmd.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

ChristophHaag, 2018-05-12:
/home/tekjanson/Documents/Personal/SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h:1538:71: error: ‘nullptr’ was not declared in this scope

Hmm... It looks like nullptr is c++11 feature and it looks like my cmake script doesn't enable c++11. I think in gcc 7 and newer c++11 or c++14 are enabled by default.

I tried enabling it in 80eae3176851886e1207d2b066ccd1ed4b019d4c, maybe that works.

If your cmake is older than 3.1 you can try replacing `set (CMAKE_CXX_STANDARD 11)` with one of those ugly things found on stack overflow.
```
macro(use_cxx11)
  if (CMAKE_VERSION VERSION_LESS "3.1")
    if (CMAKE_CXX_COMPILER_ID STREQUAL "GNU")
      set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=gnu++11")
    endif ()
  else ()
    set (CMAKE_CXX_STANDARD 11)
  endif ()
endmacro(use_cxx11)
```
https://stackoverflow.com/a/31010221

ChristophHaag, 2018-05-13:
/dev/shm/SteamVR-OpenHMD/driverlog.h:20:28: error: ‘vr’ has not been declared

No idea.

`vr` is a namespace that comes from openvr's driver header which cmake downloads as subprojects/openvr/openvr_driver.h. There is no error on the include of openvr_driver.h so it probably downloaded. Maybe check the content of that file... If it really is that file https://raw.githubusercontent.com/ValveSoftware/openvr/master/headers/openvr_driver.h then I don't know

h1z1, 2018-05-13:
Something certainly weird.  File was 0 bytes but even after a complete checkout again they don't match:

cb28a3e929dcee3e8dccb53161d8bf34  ValveSoftware/openvr/headers/openvr_driver.h
6816cd2b022275410b4b197edc9adc9d  SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h

make does however clobber it o_O after

```
ls -alh ValveSoftware/openvr/headers/openvr_driver.h SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h
-rw-rw-r-- 1 ovr ovr 115K May 13 04:58 ValveSoftware/openvr/headers/openvr_driver.h
-rw-rw-r-- 1 ov ovr    0 May 13 05:02 SteamVR-OpenHMD/subprojects/openvr/openvr_driver.h

```

ChristophHaag, 2018-05-13:
Looks like I accidentally added an old version to git instead of having it freshly downloaded by cmake. I think it never updated the file after having it downloaded it once so It wasn't a great idea anyway. 

ecfda0c5a8b0aeee4d014160829fe2061ae312d0 does away with that download and just adds the most recent version of the file to the repository.

tekjanson, 2018-05-13:
Awesome that did it thanks :)