ChristophHaag/SteamVR-OpenHMD #5: seems vrcompositor hangs on a black screen when using steamVR-openHMD!

hradec, 2018-08-19:
Hi there... 

Here I am again on my fight to make steamVR work!! LOL

I finally got the vrcompositor to run successfully using the NULL driver from valve, so FINALLY steamVR is running as it should on my arch distro. 

Now that it runnning 100% with the null driver, I known the problem is related to this openhmd driver. SteamVR is loading it successfully, showing the HMD icon exactly like yours in the video on YouTube. 

BUT, the vrcompositor window is black, and I get a "(unresposive) SteamVR Home" message on the Status window. 

After a few minutes, that message disappears and I get a beautiful green "Ready" message instead, but the vrcompositor is still black. 
I also get this on the shell at the same time: 
```
CAppInfoCacheReadFromDiskThread took 1 milliseconds to initialize
CApplicationManagerPopulateThread took 78 milliseconds to initialize (will have waited on CAppInfoCacheReadFromDiskThread)
Looking up breakpad interfaces from steamclient
Calling BreakpadMiniDumpSystemInit
Steam_SetMinidumpSteamID:  Caching Steam ID:  76561197993337696 [API loaded yes]
Steam_SetMinidumpSteamID:  Setting Steam ID:  76561197993337696
assert_20180818232128_1.dmp[21812]: Uploading dump (out-of-process)
/tmp/dumps/assert_20180818232128_1.dmp
assert_20180818232128_1.dmp[21812]: Finished uploading minidump (out-of-process): success = yes
assert_20180818232128_1.dmp[21812]: response: CrashID=bp-d621494e-f1b0-48af-a78c-b4d282180818
assert_20180818232128_1.dmp[21812]: file ''/tmp/dumps/assert_20180818232128_1.dmp'', upload yes: ''CrashID=bp-d621494e-f1b0-48af-a78c-b4d282180818''
/ssd/dot_local_share_Steam/Steam/steamapps/common/SteamVR/tools/steamvr_environments/game/steamtours.sh: line 94: 20532 Segmentation fault      (core dumped) ${STEAM_RUNTIME_PREFIX} ${GAME_DEBUGGER} "${GAMEROOT}"/${GAMEEXE} "$@"

```
The "Frame Timing" window doesn't show anything running either... it's like something is "stuck" during starting up, and can't continue!

Again, with the NULL driver, this doesn't happen... so it must be something related to openHMD or maybe some missing configuration.

I couldn't find anything wrong in the log files, but I'll keep examining then... 

My question is: Have you seem this problem before? any insights you can think off?

hradec, 2018-08-19:
I was actually wondering if this behavior has anything to do with controller bindings... I've tried to

ChristophHaag, 2018-08-19:
I have not seen this and don't know what could cause it. Sorry.

hradec, 2018-08-20:
One more question... did you setup any special bindings on your setup? I notice there are no bindings and vrcompositor reverts to a default binding file that also has nothing in it.

ChristophHaag, 2018-08-20:
Do you mean action bindings for the controller buttons? That's missing because I didn't get the button input to work yet. I've tried a bit in a branch called experiments, but I think it's missing something. I don't have nolo controllers currently so I don't have hardware for testing this. I thought I wait for the Vive controller support at least with rotation only before playing more with it, but I could also just simulate some buttons.