ChristophHaag/SteamVR-OpenHMD #6: Input Issues.

minake14, 2018-09-17:
So, I've been messing around with this project and your simulator driver, everything compiled, SteamVR launches, and the GTK input window opens. However, when I move the sliders, SteamVR doesn't register the input from the simulator driver. Everything works fine when I use the simulator with the examples included in OpenHMD by default. I've also tried sending input from a UDP port to OpenHMD-SteamVR, but I get the same results. Any ideas?

Any assistance would be greatly appreciated!

ChristophHaag, 2018-09-17:
The only thing is that if the HMD is not moved for a few seconds, SteamVR goes into "standby mode" and freezes the rendering, until the HMD is moved again.

Other than that I made a rudimentary configuration file in ~/ohmd_config.txt where you have to make sure the simulated HMD and controllers are set up to be used.

If it's not that, no idea.

minake14, 2018-09-17:
Thanks a lot! I figured out what was wrong. It seems like SteamVR really dislikes communication between multiple threads. Any values that taken from the loop of the gtk thread is unable to be sent to SteamVR. Maybe it's an issue with my system.

Should I leave this open in case someone else runs into this issue?

closed 2019-07-01 by minake14