ChristophHaag/SteamVR-OpenHMD #13: Socket closed

fraunos, 2019-02-13:
Hi, setting up went good until I got this in vrserver.txt
```
Wed Feb 13 2019 23:04:14.807320 - //==============================================================================================
Wed Feb 13 2019 23:04:14.807705 - ================================================================================================
Wed Feb 13 2019 23:04:14.807718 - ================================================================================================
Wed Feb 13 2019 23:04:14.807732 - vrserver startup with PID=2306, config=/home/michalw/.local/share/Steam/config, runtime=/home/michalw/.local/share/Steam/steamapps/common/SteamVR
Wed Feb 13 2019 23:04:14.807794 - Runtime path was /home/michalw/.local/share/Steam/steamapps/common/SteamVR. Setting it to /home/michalw/.local/share/Steam/steamapps/common/SteamVR/ to match vrserver
Wed Feb 13 2019 23:04:14.809337 - Setting shared memory namespace to 2306
Wed Feb 13 2019 23:04:14.809549 - VR server (v1548804153) starting up with config=/home/michalw/.local/share/Steam/config
Wed Feb 13 2019 23:04:14.809562 - VRServer will wait for VRMonitor before exiting
Wed Feb 13 2019 23:04:14.809722 - CIPCPipe::CreatePipe(VR_ServerPipe_2306) bound to 52019
Wed Feb 13 2019 23:04:14.830730 - [Settings] Load Default Json Settings from /home/michalw/.local/share/Steam/steamapps/common/SteamVR/resources/settings/default.vrsettings
Wed Feb 13 2019 23:04:14.830871 - [Settings] Load Default Json Settings from /home/michalw/.local/share/Steam/steamapps/common/SteamVR/drivers/htc/resources/settings/default.vrsettings
Wed Feb 13 2019 23:04:14.830930 - [Settings] Load Default Json Settings from /home/michalw/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/resources/settings/default.vrsettings
Wed Feb 13 2019 23:04:14.830988 - [Settings] Load Default Json Settings from /home/michalw/vr/SteamVR-OpenHMD/build/resources/settings/default.vrsettings
Wed Feb 13 2019 23:04:14.831039 - [Settings] Load Default Json Settings from /home/michalw/.local/share/Steam/steamapps/common/SteamVR/drivers/null/resources/settings/default.vrsettings
Wed Feb 13 2019 23:04:14.831607 - [Settings] Load Json Settings from /home/michalw/.local/share/Steam/config/steamvr.vrsettings
Wed Feb 13 2019 23:04:14.831980 - Unable to read app config file from /home/michalw/.local/share/Steam/config/appconfig.json. No applications will be available
Wed Feb 13 2019 23:04:14.834929 - /home/michalw/.local/share/Steam/steamapps/common/SteamVR/tools/tools.vrmanifest - App openvr.tool.steamvr_media_player must specify binary_path for launch_type binary. Skipping
Wed Feb 13 2019 23:04:14.835125 - /home/michalw/.local/share/Steam/steamapps/common/SteamVR/tools/tools.vrmanifest - App openvr.tool.steamvr_desktop_gametheatre Working directory /home/michalw/.local/share/Steam/steamapps/common/SteamVR/tools/steamvr_desktop_game_theater/linux64 is invalid. Skipping. Skipping
Wed Feb 13 2019 23:04:14.838079 - /home/michalw/.local/share/Steam/steamapps/common/SteamVR/resources/config/runtime.vrmanifest - App openvr.component.vrwebhelper binary_path /home/michalw/.local/share/Steam/steamapps/common/SteamVR/bin/vrwebhelper/vrwebhelper is invalid. SKipping. Skipping
Wed Feb 13 2019 23:04:14.985503 - New Connect message from /home/michalw/.local/share/Steam/steamapps/common/SteamVR/bin/linux64/vrstartup (VRApplication_Bootstrapper) (Args: ) 2299 
Wed Feb 13 2019 23:04:14.985705 - Add Json firmware manifest from {htc}/firmware/manifest.vrfirmware
Wed Feb 13 2019 23:04:14.985926 - Setting app openvr.component.vrstartup PID to 2299
Wed Feb 13 2019 23:04:14.986369 - Add Json firmware manifest from {lighthouse}/firmware/manifest.vrfirmware
Wed Feb 13 2019 23:04:14.986605 - [Web] Starting web server on port localhost:8998
Wed Feb 13 2019 23:04:14.990614 -   driver gamepad implements interfaces IVRSettings_002 ITrackedDeviceServerDriver_005 IVRDisplayComponent_002 IVRDriverDirectModeComponent_005 IVRCameraComponent_003 IServerTrackedDeviceProvider_004 IVRWatchdogProvider_001 IVRVirtualDisplay_002 IVRDriverManager_001 IVRResources_001 IVRCompositorPluginProvider_001 
Wed Feb 13 2019 23:04:14.990650 - Loaded server driver gamepad (IServerTrackedDeviceProvider_004) from /home/michalw/.local/share/Steam/steamapps/common/SteamVR/drivers/gamepad/bin/linux64/driver_gamepad.so
Wed Feb 13 2019 23:04:14.990670 - Driver gamepad has no suitable devices.
Wed Feb 13 2019 23:04:14.990686 - Not loading driver lighthouse because it is disabled in settings
Wed Feb 13 2019 23:04:14.990699 - Not loading driver oculus because it is disabled in settings
Wed Feb 13 2019 23:04:14.990711 - Not loading driver oculus_legacy because it is disabled in settings
Wed Feb 13 2019 23:04:15.007631 - openhmd: device 0
Wed Feb 13 2019 23:04:15.007659 - openhmd:   vendor:  Sony
Wed Feb 13 2019 23:04:15.007667 - openhmd:   product: PSVR
Wed Feb 13 2019 23:04:15.007677 - openhmd:   path:    0

Wed Feb 13 2019 23:04:15.007684 - openhmd: device 1
Wed Feb 13 2019 23:04:15.007691 - openhmd:   vendor:  OpenHMD
Wed Feb 13 2019 23:04:15.007698 - openhmd:   product: External Device
Wed Feb 13 2019 23:04:15.007705 - openhmd:   path:    (none)

Wed Feb 13 2019 23:04:15.007711 - openhmd: device 2
Wed Feb 13 2019 23:04:15.007718 - openhmd:   vendor:  OpenHMD
Wed Feb 13 2019 23:04:15.007725 - openhmd:   product: HMD Null Device
Wed Feb 13 2019 23:04:15.007731 - openhmd:   path:    (none)

Wed Feb 13 2019 23:04:15.007738 - openhmd: device 3
Wed Feb 13 2019 23:04:15.007745 - openhmd:   vendor:  OpenHMD
Wed Feb 13 2019 23:04:15.007752 - openhmd:   product: Left Controller Null Device
Wed Feb 13 2019 23:04:15.007759 - openhmd:   path:    (none)

Wed Feb 13 2019 23:04:15.007776 - openhmd: device 4
Wed Feb 13 2019 23:04:15.007784 - openhmd:   vendor:  OpenHMD
Wed Feb 13 2019 23:04:15.007791 - openhmd:   product: Right Controller Null Device
Wed Feb 13 2019 23:04:15.007798 - openhmd:   path:    (none)

Wed Feb 13 2019 23:04:15.007815 - openhmd: could not open config file /home/michalw/.ohmd_config.txt, using default headset 0 with no controller
Wed Feb 13 2019 23:04:15.007827 - openhmd: could not open config file /home/michalw/.ohmd_config.txt, using default headset 0 with no controller
Wed Feb 13 2019 23:04:15.007836 - openhmd: could not open config file /home/michalw/.ohmd_config.txt, using default headset 0 with no controller
Wed Feb 13 2019 23:04:15.007846 - openhmd: could not open config file /home/michalw/.ohmd_config.txt, using default headset 0 with no controller
Wed Feb 13 2019 23:04:15.007854 - openhmd: Using HMD Display 0, HMD Tracker 0, Left Controller -1, Right Controller -1
Wed Feb 13 2019 23:04:15.021196 - openhmd: Using settings values
Wed Feb 13 2019 23:04:15.021277 - openhmd: driver_openhmd: Serial Number: PSVR: 0
Wed Feb 13 2019 23:04:15.021291 - openhmd: driver_openhmd: Model Number: OpenHMD: PSVR
Wed Feb 13 2019 23:04:15.021303 - openhmd: driver_openhmd: Window: 1920 0 1920 1080
Wed Feb 13 2019 23:04:15.021314 - openhmd: driver_openhmd: Render Target: 1920 1080
Wed Feb 13 2019 23:04:15.021329 - openhmd: driver_openhmd: Seconds from Vsync to Photons: 0.011000
Wed Feb 13 2019 23:04:15.021341 - openhmd: driver_openhmd: Display Frequency: 0.000000
Wed Feb 13 2019 23:04:15.021353 - openhmd: driver_openhmd: IPD: 0.061000
[... 7 lines cut ...]
Wed Feb 13 2019 23:04:15.022565 - openhmd: projectionraw values lrtb, near far: -1.130760 1.127176 -1.270089 1.270089 | 0.100000 1000.530334
Wed Feb 13 2019 23:04:15.022583 - openhmd: projectionraw values lrtb, near far: -1.127176 1.130760 -1.270089 1.270089 | 0.100000 1000.530334
Wed Feb 13 2019 23:04:15.022596 - Guessing frequency of current display mode because SDL failed us
Wed Feb 13 2019 23:04:15.022611 - Initialized EyeToHead Left Translation -0.030500 0.000000 0.000000 
Wed Feb 13 2019 23:04:15.022624 - Initialized EyeToHead Right Translation 0.030500 0.000000 0.000000 
Wed Feb 13 2019 23:04:15.023429 - Adding controller type to openhmd_hmd
Wed Feb 13 2019 23:04:15.023743 - Driver 'openhmd' finished adding tracked device with serial number 'PSVR: 0'
Wed Feb 13 2019 23:04:15.023910 - Processing message VRMsg_Connect from vrstartup (2299) took 0.0385 seconds
Wed Feb 13 2019 23:04:15.023998 - Failed to read chaperone file from /home/michalw/.local/share/Steam/config/chaperone_info.vrchap
Wed Feb 13 2019 23:04:15.104421 - New Connect message from /home/michalw/.local/share/Steam/steamapps/common/SteamVR/bin/linux64/vrcompositor (VRApplication_Background) (Args: ) 2331 
[... cut ...]
Wed Feb 13 2019 23:04:16.864410 - Setting app openvr.tool.steamvr_environments PID to 2365
Wed Feb 13 2019 23:04:18.487280 - Socket closed
Wed Feb 13 2019 23:04:18.487358 - Unable to read message from socket: 17
Wed Feb 13 2019 23:04:18.487411 - Process vrcompositor (2331) disconnected (Thread(0x0x7f6bb0002ba0/0x0x7f)
Wed Feb 13 2019 23:04:18.487447 - Clearing application openvr.component.vrcompositor PID because 2331 has exited
Wed Feb 13 2019 23:04:18.872930 - IPC: recovering abandoned mutex 0x7f6bd99d3018
```

Is this bad OpenHMD setup or something else? I got no ideas.

fraunos, 2019-02-13:
I get a red screen and it crashes the box. Just to add, I got AMD GPU, but opengl example from OpenHMD works.

minake14, 2019-02-15:
Hi, I'm not the dev, but I do use this project on an almost daily basis and it works well. Couple questions:
Are you using the Steam beta client? 
Are you using the SteamVR beta client?
Did you test SteamVR was working before you setup SteamVR-OpenHMD by enabling the null driver?

fraunos, 2019-02-15:
@minake14
Um, nope, nope, nope? How to do the third? Thanks, I expected I missed something. Gonna try this out this evening.

minake14, 2019-02-15:
No problem. First, get yourself into both of the betas, then try and launch it as is. If that does not work, go into the SteamVR-OpenHMD and edit this part of "steamvr.vrsettings":

"driver_openhmd" : {
		 "enable": true
   },

Just change "true" to "false" and save. After that, browse the local files of SteamVR to find  the default.vrsettings for the null driver and edit it so that "enable" is true. Try to launch it again. This is test whether or not it's SteamVR or the OpenHMD driver that's causing the issue. If it works, it's OpenHMD. If it doesn't, it's SteamVR. Usually it's SteamVR  

The path to the null driver should look something like this: Steam/steamapps/common/SteamVR/drivers/null/resources/settings.

fraunos, 2019-02-15:
Yeah, `Shared IPC Compositor Invalid Connect Response (307)`

minake14, 2019-02-16:
That was with the null driver test? Also what distro and version of mesa are you running? I'm running Debian Buster(testing) and mesa 18.2 with LLVM 7 on an RX580.

fraunos, 2019-02-16:
Right, maybe I should write on SteamVR Issues? Looks like @hradec had similar issue https://github.com/ValveSoftware/steam-for-linux/issues/4475

Anyway:
#### Your system information

* Steam client version (build number or date): 15.02.2019
* Distribution (e.g. Ubuntu):  Manjaro (I tried both steam-manjaro and steam-native packages)
* Graphics driver version : RX 470 with Mesa 18.3.2-1
* Gist for SteamVR System Information: https://gist.github.com/fraunos/3f63193dc8b47773c984b73917f73163
* Opted into Steam client beta?: [Yes/No] Yes
* Opted into SteamVR beta?: [Yes/No] Yes
* Have you checked for system updates?: [Yes/No] Yes

I tried:
https://bbs.archlinux.org/viewtopic.php?pid=1770146#p1770146
https://aur.archlinux.org/packages/psvr-udev/
No success

minake14, 2019-02-16:
Yeah, sorry I couldn't help.

The last thing I can think of is maybe you don't have the Vulkan loader installed. Maybe testing a Vulkan game would help. Otherwise, it would be best to open an issue here:

https://github.com/ValveSoftware/SteamVR-for-Linux/issues

Here's the issue template:  

https://github.com/ValveSoftware/SteamVR-for-Linux/blob/master/issue_template.md

You either tell them what you told me or just like them to this thread. Good luck!

fraunos, 2019-02-16:
Thanks! I play Dota 2 with Vulkan, so I guess Vulkan is working :vulcan_salute:

ChristophHaag, 2019-02-18:
Sorry, just saw this issue now.

I think the log in the first post looks good, but

> I get a red screen and it crashes the box.

doesn't sound good.

After one of these system crashes you can run `sudo journalctl -k -b -1` to show the **k**ernel log from **1** **b**oot in the past and look at the end if there are some messages from the amdgpu module, perhaps a GPU hang? In that case it would unfortunately be a driver issue, though Polaris (RX 4xx and 5xx) should be well tested and not really have those issues...

fraunos, 2019-02-19:
Hi, sorry for not specifying - I meant PSVR box, it's fan worked loudly and all next runs ended up with nothing, showing up at all. I'm just gonna post it on SteamVR.