ChristophHaag/SteamVR-OpenHMD #14: Standard SteamVR folder moved

jug007, 2019-02-18:
The standard Steam installation folder has been move (at least on ubuntu), which makes the installation script of SteamVR-OpenHMD fail.
The standard folder needs to be updated and there should be an option to change the SteamVR folder to custom installation directorys.

The new standard Steam installation folder is `/home/USER/.steam/`
So SteamVR would normally be installed in: `/home/USER/.steam/steam/steamapps/common/SteamVR/`

Workaround to the problem would be to make a symlink: `ln -s /home/USER/.steam/steam/ /home/USER/.local/share/`

ChristophHaag, 2019-02-18:
I believe Steam on Archlinux I set up recently still only used ~/.local/share/Steam and I had to create the ~/.steam/steam symlink manually.

Anyway, #10 was complaining about this too and it's not actually complicated, so I finally changed it. With 3f759c5d279c9daaddbba0edb2d2f1dcd93e10b7 I hope all configurations work.

jug007, 2019-02-18:
I just ran register.sh with the symlink removed and the new commit. Works well for me. =)

closed 2019-02-18 by jug007