ChristophHaag/SteamVR-OpenHMD #15: SteamVR 301 error Oculus CV1

jug007, 2019-02-18:
I am getting a error 301 when trying to use my Oculus Rift CV1 in SteamVR with SteamVR-OpenHMD (built from git 17.02.2019) and OpenHMD 0.2.0 from the official Ubuntu Bionic repo.

I have no idea how to troubleshoot this. Is there any logs that I could make or are already made that I could take a look at?

ChristophHaag, 2019-02-18:
SteamVR-OpenHMD should not use Ubuntu's OpenHMD, but instead a recent git revision of OpenHMD located in subprojects/openhmd, though I have not really tested if this really always happens.

Anyway, SteamVR error codes [are notoriously unhelpful](https://github.com/ValveSoftware/openvr/wiki/HmdError).

Real log messages can be found in

* ~/.steam/steam/logs/vrcompositor.txt
* ~/.steam/steam/logs/vrserver.txt

and some other log files there that start with vr

jug007, 2019-02-18:
The [vrserver.txt](https://gist.github.com/jug007/9f8ff58ae7ff7e74bb1bdeea604e7457#file-vrserver-txt) seems pretty helpful actually.
[vrmonitor.txt](https://gist.github.com/jug007/9f8ff58ae7ff7e74bb1bdeea604e7457#file-vrmonitor-txt) doesn't show anything that I would get anything out of.
vrcompositor.txt does not exist for me.

I know that it isn't always easy to have applications grab a usb device. So I guess I will need to take a closer look at that.

ChristophHaag, 2019-02-18:
Right, you need permission to open the raw usb device and by default linux distributions don't give that to users.

It's probably enough creating a file /etc/udev/rules.d/83-hmd.rules with content

    SUBSYSTEM=="usb", ATTR{idVendor}=="2833", MODE="0666", GROUP="plugdev"'

and then running `udevadm control --reload-rules` and - maybe - plugging in the headset again or rebooting.

jug007, 2019-02-18:
Omg it worked.
After turning on direct mode in SteamVR I get rotational tracking, sound and image on my rift. The image is distorted, but I guess that is an OpenHMD problem.
Now I just gotta figure out how to change the HMD position in SteamVR up from the floor and I can actually play games on this. =3

ChristophHaag, 2019-02-18:
Fun fact: On Linux you can't turn on or off direct mode. If the graphics driver supports it, SteamVR will use it, and if the graphics driver does not, then it will fall back to extended mode. The direct mode button in the SteamVR doesn't do anything on Linux. Glad to hear that direct mode still automagically works though.

The SteamVR Room Setup should take care of the HMD height positioning, but I have raised that point at OpenHMD too because HMDs that OpenHMD only supports with rotation tracking report only (0,0,0) as position, which is "stuck on the floor" so applications (in this case SteamVR) have to translate that to a useful position. Having a default of 1.6-1.8 meter for only rotationally tracked HMDs might be a good idea.

OpenHMD should have relatively good distortion correction for Rift DK1, DK2 and CV1 and SteamVR-OpenHMD passes this distortion correction to SteamVR. I haven't tried SteamVR-OpenHMD myself with a Rift though, so no idea if something is wrong.

jug007, 2019-02-18:
So yeah if it switched back to extended mode I get no image on the oculus.
The height also works now.
And yeah if there is distrotion correction, it is either very bad or not working for some reason.
Still really cool to be able to have the Rift somewhat working.

closed 2019-02-18 by jug007