ChristophHaag/SteamVR-OpenHMD #21: Getting gaze cursor working

interfect, 2019-03-09:
I'm trying to use SteamVR-OpenHMD with just an Oculus DK1 headset and a gamepad, on Ubuntu.

I want to be able to operate the menus in SteamVR Home.

I am able to get everything hooked up, and look around in SteamVR home, and even jump around by hitting one of the right shoulder buttons and looking at where I want to go. Head rotation tracking is fine, and the rendering/distortion seems fine as well.

My problem is that I can't hit any of the buttons. When I load in, the wall in front of me says to press Y to open the menu. When I do that, a floating menu comes up in front of me, with a button on it that it says I am supposed to activate with "Trigger". I can't do anything to get the button to click.

I've also tried [SteamVR-OSVR](https://github.com/OSVR/SteamVR-OSVR) with the same hardware. When I do that, I seem to have some sort of gaze control/pointer that I don;t have with SteamVR-OpenHMD. When I get the same menu, I can look at the button it wants me to click and hit the A-button-equivalent on my controller to click the button. I also get little extra menus coming up when I hover my gaze at the side of that floating menu, and when I bring up the teleport/jump overlay with that shoulder button, I have a green cursor that moves wherever I look on the floor, to show me where I am going to teleport to. Unfortunately, SteamVR-OSVR can't handle rendering correctly; the images for the two eyes that I get don't come to a coherent 3D view, and everything weirdly changes size in the vertical and horizontal dimensions as I tilt my head. Plus, the whole OSVR ecosystem is both much more complex than OpenHMD and also apparently abandoned, so I would much rather use this project.

Does anyone have any insight into what SteamVR-OSVR might be sending in to SteamVR to make its gaze cursor work? And how SteamVR-OpenHMD could be modified to do the same thing?

My config file looks like this:

```
hmddisplay 0
hmdtracker 0
leftcontroller -1
rightcontroller -1
```

I haven't tried telling it that my head is one of the controllers. Perhaps that would work?

ChristophHaag, 2019-03-09:
Interesting. I tried with SteamVR's Vive driver, Vive Controllers powered off, XBox 360 gamepad connected and do not see gaze controls like you describe.

Perhaps the plugin needs to tell SteamVR that it does not support controllers at all for it to be activated or some other special code is needed.

interfect, 2019-03-10:
Here's a bunch of stuff that doesn't help:
* Using `leftcontroller 0` in the config file.
* Changing the reported interface set from `IVRSettings_002 ITrackedDeviceServerDriver_005 IVRDisplayComponent_002 IVRDriverDirectModeComponent_005 IVRCameraComponent_002 IServerTrackedDeviceProvider_004 IVRWatchdogProvider_001 IVRVirtualDisplay_002 IVRDriverManager_001 IVRResources_001` to `IVRSettings_002 ITrackedDeviceServerDriver_005 IVRDisplayComponent_002 IVRDriverDirectModeComponent_002 IVRControllerComponent_001 IVRCameraComponent_002 IServerTrackedDeviceProvider_004 IVRWatchdogProvider_001 IVRVirtualDisplay_001` like OSVR reports.
* Adding a `vr::TrackedDeviceClass_TrackingReference` device that just sits there, like OSVR does.
* Setting a nonzero `poseTimeOffset` in the HMD pose.
* Setting `willDriftInYaw` and `shouldApplyHeadModel` in the pose.
* Normalizing the pose head rotation quaternion.
* Moving HMD tracking and calling TrackedDevicePoseUpdated into a thread.
* Adding `ITrackedDeviceServerDriver` as a supported component in the HMD driver's `GetComponent`. 

Copying *all* the properties that OSVR sets over seems like it may have gotten it working. I'm currently trying to prune that down to just whatever is actually required.

ChristophHaag, 2019-03-10:
Thanks for looking into this, this will be quite useful.

And yes, SteamVR-OpenHMD is not very complete, I didn't do a lot of work on it after it started working, partly because it's quite painful to work with such little documentation...

closed 2019-03-11 by ChristophHaag

ChristophHaag, 2019-03-11:
Thanks for investigating, I added an environment variable to override the manufacturer string and tried with my Vive and OpenHMD and SteamVR Home kind of works.

PR #22 (2019-03-11): Fix gaze cursor for Oculus devices
SteamVR checks if Prop_ManufacturerName_String == "Oculus" somewhere in their
UI code. If you don't send it, and you don't have any real controllers, you
don't get the fallback gaze cursor, so you can't use the menus.

I'm grabbing the first word of OHMD_VENDOR to provide the manufacturer name,
which is "Oculus" on the Oculus DK1 that I have.

This lets me get a gaze cursor and navigate SteamVR Home with the headset and a
gamepad.

I've also replaced some fragile-looking strcpy code with std::stringstream.

This should fix #21.
commit 98b064fe (2019-03-11): Fix gaze cursor for Oculus devices

SteamVR checks if Prop_ManufacturerName_String == "Oculus" somewhere in their
UI code. If you don't send it, and you don't have any real controllers, you
don't get the fallback gaze cursor, so you can't use the menus.

I'm grabbing the first word of OHMD_VENDOR to provide the manufacturer name,
which is "Oculus" on the Oculus DK1 that I have.

This lets me get a gaze cursor and navigate SteamVR Home with the headset and a
gamepad.

I've also replaced some fragile-looking strcpy code with std::stringstream.

This should fix #21.