ChristophHaag/SteamVR-OpenHMD #24: Oculus CV1 image distortion

jug007, 2019-04-02:
As I haven't been able to get any more information on this issue, I will just open this Issue here without much more information.

Using the Oculus Rift CV1 with SteamVR in Direct Mode has the image distorted.
SteamVR-OpenHMD reports distortion parameters being set in the [log](https://gist.github.com/jug007/1a0827900558d2939975396c3b841565). According to the OpenHMD source code, these are the right parameters for the Rift.

I haven't found an application than I can try direct mode with and I don't know how you are supposed to use any application with extended mode.

The distrotion isn't gamebreaking, but clearly visible, so I am not even sure if the distortion values that come with OpenHMD are right.

jug007, 2019-06-11:
Just set up a bounty for who ever manages to get this fixed, as this is somewhat important to me.
https://www.bountysource.com/issues/72293469-oculus-cv1-image-distortion

ChristophHaag, 2019-06-11:
> I haven't found an application than I can try direct mode with and I don't know how you are supposed to use any application with extended mode.

Hey. In the meantime I wrote down a little bit of what I know about extended mode in the OpenHMD wiki: https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode

Specifically for the CV1 someone made an issue and it turns out that the display only works while an OpenHMD app (or SteamVR-OpenHMD) is running. https://github.com/OpenHMD/OpenHMD/issues/224

Unfortunately I still can't say anything about the Oculus Rift distortion, because I don't have one.

Just a suggestion, but if you are interested in testing more stuff, you can try our (Collabora) OpenXR runtime which supports direct mode and among others uses OpenHMD drivers: https://gitlab.freedesktop.org/monado/monado ([simple compilation instructions](https://gitlab.freedesktop.org/monado/monado/wikis/building-without-installing))
Though right now there are only few OpenXR examples and no real applications you can run on it right now.

jug007, 2019-06-11:
The display turns on no problem for me in extended mode. I just have no idea how I am going to get an application to fullscreen on that.
I did https://gitlab.com/lubosz/xrgears before, though I couldn't get it to run.
If your are up for it and maybe not too far away, I could lend you my Oculus to take a look yourself.

Szalacinski, 2019-06-20:
The distortion is very slight, correct?  Do objects appear slightly smaller when at the center?

jug007, 2019-06-21:
I wouldn't call it *very* slight. Thought it is not very bad either. I haven't really noticed objects being smaller in the center (will look out for that when I have the headset on next). So the distrortion is so bad that I have had dizzyness and headaches when I started using the oculus with this. I have gotten used to it though, to the point where I only have problems with orientation (and immersion).
What I specificly notice is that staight lines are not straight in the oculus anymore. They seem straight in the center area, but warp shortly after getting further out.
I guess I should *try* to take a picture of it, so people can actually see what I am talking about.

jug007, 2019-06-21:
I tried my best to take some pictures:
[image]
[image]
[image]

You can clearly see everything being somewhat warped.
Here a (weirdly squished) screenshot made with SteamVR for comparison:
[image]
It might be squished, but you can still see that all the edges are supposed to be straight, rather than warped.

gadLinux, 2020-06-12:
I can confirm this issue. I use to play EliteDangerous and I cannot because distortion. 
It seems that something is doing wrong in the center. It's like a ripple and I can see it when I move the head. Just strange. It happens on SteamVR at all times on direct mode.

gadLinux, 2020-08-01:
I managed to make a couple of videos

http://owncloud.level2crm.com/index.php/s/oGkd5ZJLbfqQ6Ag
http://owncloud.level2crm.com/index.php/s/kLwqy6tKnwMf7LE

I hope you can see it. In stereo the effect is bigger...

thaytan, 2021-09-01:
I just fixed CV1 lens distortion parameters in my positional tracking branch. See https://github.com/thaytan/OpenHMD/issues/22

JulianGro, 2021-09-01:
Unfortunately I don't have a Rift to test this on anymore. But I guess this can still be closed once the new parameters make it into upstream.