ChristophHaag/SteamVR-OpenHMD #28: [AMD] Displays SteamVR content to monitor and not to oculus headset

Doomsdayrs, 2019-06-26:
Similar to #27 But the issue is on an AMD card (RX 580).

Doomsdayrs, 2019-06-26:
Added 'sleep(1);' to line 408 of rift.c fixes it

Shoobledoops, 2020-06-17:
Hi there,

I seem to be having the same issue as you. Could you provide a bit more information on how you fixed it? The location of the rift.c file, and a copy of the specific code block would be a huge help. :)

djibux, 2020-11-29:
`rift.c` is in folder `SteamVR-OpenHMD/subprojects/openhmd/src/drv_oculus_rift`.
When the comment was submitted, [line 408](https://github.com/OpenHMD/OpenHMD/blob/179553d301ded314a144051510cf5f7d471fc7d2/src/drv_oculus_rift/rift.c#L408) was the last but one line in the `open_device` function.  So I guess that's where you should put it.
You also need to add `#include <unistd.h>` with the other `include` directives.

djibux, 2020-11-29:
I should mention that I have the same issue with the Rift S.  The fix above didn't work (I tried adapting it to `drv_oculus_rift_s`).

To get it to work I need to:
* Start SteamVR : it displays on my main monitor, but head tracking works
* Set up [Extended Mode](https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode), using the command `xrandr --output DisplayPort-0 --set non-desktop 0`
* Move SteamVR to the helmet now configured as a desktop (I use _Shift+Meta+Right Arrow_ on Gnome).

It works OK, but head tracking is jumpy.  I don't know if performance would be better with Direct Mode.

Odisej75, 2020-12-28:
Having the same problem with rift s + AMD (rx590). 

Does anybody have a clue where to start searching for the reasons?