ChristophHaag/SteamVR-OpenHMD #30: Add documentation on how to uninstall OpenHMD from SteamVR completely

jug007, 2019-07-29:
It actually seems really had to properly remove OpenHMD from SteamVR once installed.
It took me almost 4 hours to get SteamVR to a point where I was able to use my Valve Index instead and I still get error messages saying that OpenHMD was not found.

Reinstalling SteamVR and/or removing its configs does not seem enough.

ChristophHaag, 2019-07-29:
That sucks.

I'm not quite sure where SteamVR stores the config, but the `SteamVR/bin/linux64/vrpathreg` tool is used to register and remove drivers. If you run it without arguments it should print the registered drivers under "External Drivers" like this:
```
LD_LIBRARY_PATH=~/.local/share/steam/steamapps/common/SteamVR/bin/linux64 ~/.local/share/Steam/steamapps/common/SteamVR/bin/linux64/vrpathreg
Runtime path = /home/chris/.local/share/Steam/steamapps/common/SteamVR
Config path = /home/chris/.local/share/Steam/config/
Log path = /home/chris/.local/share/Steam/logs/
External Drivers:
        /home/chris/SteamVR-OpenHMD/build
```

If you have libopenvr_api.so in the library path like /usr/lib/libopenvr_api.so, then you don't need LD_LIBRARY_PATH.

Depending on the distribution package Steam was installed from it might be ~/.steam/steam instead of ~/.local/share/Steam.

Something like this should remove the driver from SteamVR:
```
LD_LIBRARY_PATH=~/.local/share/steam/steamapps/common/SteamVR/bin/linux64 ~/.local/share/Steam/steamapps/common/SteamVR/bin/linux64/vrpathreg removedriver  /home/chris/SteamVR-OpenHMD/build
```

That's what the unregister.sh script is trying to do.

jug007, 2019-07-29:
Worked fine.
```
motofckr9k@MotoFckr9k-Tessa:~/.steam/steam/steamapps/common/SteamVR/bin/linux64$ LD_LIBRARY_PATH=~/.steam/steam/steamapps/common/SteamVR/bin/linux64 ~/.steam/steam/steamapps/common/SteamVR/bin/linux64/vrpathreg
Runtime path = /home/motofckr9k/.steam/steam/steamapps/common/SteamVR
Config path = /home/motofckr9k/.steam/steam/config
Log path = /home/motofckr9k/.steam/steam/logs
External Drivers:
	/home/motofckr9k/Dokumente/SteamVR-OpenHMD/build
motofckr9k@MotoFckr9k-Tessa:~/.steam/steam/steamapps/common/SteamVR/bin/linux64$ 
```
```
motofckr9k@MotoFckr9k-Tessa:~/.steam/steam/steamapps/common/SteamVR/bin/linux64$ LD_LIBRARY_PATH=~/.steam/steam/steamapps/common/SteamVR/bin/linux64 ~/.steam/steam/steamapps/common/SteamVR/bin/linux64/vrpathreg removedriver  /home/motofckr9k/Dokumente/SteamVR-OpenHMD/build
motofckr9k@MotoFckr9k-Tessa:~/.steam/steam/steamapps/common/SteamVR/bin/linux64$
```
```
motofckr9k@MotoFckr9k-Tessa:~/.steam/steam/steamapps/common/SteamVR/bin/linux64$ LD_LIBRARY_PATH=~/.steam/steam/steamapps/common/SteamVR/bin/linux64 ~/.steam/steam/steamapps/common/SteamVR/bin/linux64/vrpathreg
Runtime path = /home/motofckr9k/.steam/steam/steamapps/common/SteamVR
Config path = /home/motofckr9k/.steam/steam/config
Log path = /home/motofckr9k/.steam/steam/logs
motofckr9k@MotoFckr9k-Tessa:~/.steam/steam/steamapps/common/SteamVR/bin/linux64$ 
```
I will add it to the readme and make a pull request.

PR #31 (2019-07-30): Added Remove section to the Readme
Adds examples on how to remove SteamVR-OpenHMD from the vrpathregistry.
Closes #30
closed 2019-07-30 by ChristophHaag