ChristophHaag/SteamVR-OpenHMD #32: Question -- what would it take to get controller buttons working?

steamp0rt, 2019-08-17:
Out of curiosity, what would it take? Reverse engineering, trying to bind things properly, trial/error, or something else?

ChristophHaag, 2019-08-17:
Mostly it would take knowing how SteamVR expects drivers to give input data to SteamVR. At the time I was looking at it, there was no example code and documentation was not enough for me to get it working, but I could give it another try.

Also there is this branch that makes the Vive controllers work as 3dof controllers, which would give me input data from a real world device which would make it more interesting to develop. https://github.com/lubosz/OpenHMD/commits/wip/vive-controller

So I may take another look soon, but I'd also be happy for anyone to contribute some code that does it. :)

ChristophHaag, 2019-08-17:
#12 would be the starting point.

I think first I need to write a demo application that uses the SteamVR Input System instead of the legacy input so we can properly test it.

steamp0rt, 2019-08-17:
I could test on the Rift CV1

steamp0rt, 2019-08-21:
because if this worked fully i'd be able to switch to linux fully :D

minake14, 2019-08-22:
> I could test on the Rift CV1

Have you managed to get your CV1 touch controllers working with OpenHMD alone? At the time I created the pull request, I couldn't get the CV1 touch controllers working, but I tested it with @ChristophHaag's simulator and almost all the buttons I tried to implement worked.

ChristophHaag, 2020-02-28:
For my own reference: hydra driver fork, a branch with the interesting name steamvr_input_new: https://github.com/betavr/steamvr_driver_hydra/tree/r/steamvr_input_new

There has been a complaint that it doesn't work anymore though https://github.com/ValveSoftware/driver_hydra/issues/9

r57zone, 2020-05-09:
The BetaVR driver problem was in the unique ID (razer_hydra), the games did not understand it and therefore nothing worked. It was necessary to [change the ID to "vive_controller" and bind the buttons](https://github.com/r57zone/Razer-Hydra-SteamVR-driver/blob/master/drivers/driver_hydra/driver_hydra.cpp#L373-L408), after which everything worked fine.

thaytan, 2020-05-10:
I have some of the controller buttons working in the https://github.com/ChristophHaag/SteamVR-OpenHMD/tree/controller-input branch, but controller rotation / positioning was quite messed up when I tried it last.

r57zone, 2020-05-10:
I also have [problems with rotation](https://github.com/TrueOpenVR/SteamVR-TrueOpenVR/blob/master/OpenVR/samples/driver_sample/driver_sample.cpp#L692), in my [other project](https://github.com/TrueOpenVR) (DIY VR), one Euler axis uses 180 degrees, and the other two use 360 degrees, and after converting to a quaternion, problems start.

thaytan, 2020-05-10:
I need to try it again and remind myself what was wrong when I looked at it. I just remember the the controllers acted in a way I didn't expect.

witty91, 2020-07-22:
Hey, I decided to try out the controller-input branch on my rift s yesterday and again today. From what I can tell so far the triggers, B and Y are working...is that to be expected? in that case I could try to figure out what to do about the other buttons, but I have yet to understand what part steamvr-openhmd and openhmd itself are playing. Maybe someone could shed some light on that for me?

thaytan, 2020-07-22:
OpenHMD reports events for all the buttons, triggers and grips - I don't think there's anything to do in OpenHMD right now, and in the controller-input branch they all get reported to SteamVR too. What's missing is knowledge and glue in getting the input bindings set up right so that the controls actually work.

TBH - I'm surprised you got trigger, B & Y working! What was running in SteamVR then? SteamVR Home, or something else?

witty91, 2020-07-23:
I forgot to mention the menu and oculus button worked too :-D I had them working in SteamVR Home and Skyrim VR.
I didn't even configure anything yet, they just worked out of the box...maybe they picked up some other controller profile. The only thing I did was to create a 
[.ohmd_config.txt](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/4968019/default.ohmd_config.txt)
which gave me the rotations of the controllers. One thing I noticed was, that SkyrimVR was showing renders of a different VR controller that I didn't recognize...so the controllers might also have been picked up as a different controller due to my config maybe?
I'll see if I can get the rest of the controller to work next.

thaytan, 2020-07-23:
It's also possible that someone magical changed in SteamVR since I last tested :)

witty91, 2020-07-24:
Turns out ~/.ohmd_config.txt wasn't necessary...it seems to work just fine without.
I have exported the controller profile from windows and tried to copy some of the bindings from there into a profile on linux under ~/.local/share/Steam/userdata/*someuserid*/250820/remote/binding_config.json, however without much success.
Going into the settings->controller->test controller did however yield some information at least:  clicking all buttons, trigger and grip are being recognized by SteamVR. (touching the buttons doesn't do anything though). Then I tried calibrating the thumbsticks and they appear to only be recognized within the top right quadrant. (I guess that's got something to do with radians/degree conversion).
So that's where I am at currently, not sure what to look at and try next yet :-/

thaytan, 2020-07-24:
Touching the buttons doesn't work because OpenHMD doesn't have report types for capsense yet. The Rift driver can see the touch information, but can't report it to the SteamVR-OpenHMD layer.

I thought the thumbstick reports were working - but I guess not. It's been a while since I've tested that.

thaytan, 2020-07-24:
I think the missing piece here is anyone who understands how the input profile bindings stuff works. The buttons/grips/trigger reports are there... but SteamVR doesn't know how to map them to the application inputs.

witty91, 2020-07-25:
> Touching the buttons doesn't work because OpenHMD doesn't have report types for capsense yet. The Rift driver can see the touch information, but can't report it to the SteamVR-OpenHMD layer.

Ah, makes sense, just thought I should note anything I am not sure about.

> I thought the thumbstick reports were working - but I guess not. It's been a while since I've tested that.

I downloaded your rift-s-experiments repo today to see what is actually coming out of the device, and the data is certainly there.

Also, I managed to bind teleporting around to the X key, which at least gives me some mobility withing SteamVR Home! I'll proceed in creating some bindings and try to stay reasonably close to the windows ones. I think I should be able to come up with something useful either later tonight or tomorrow.

witty91, 2020-09-01:
Hi, I wanted to give a short update und this, since I have been monkeying around with these bindings. As far as I currently know, there's two ways to get bindings set up:
1) Manually Inputting them through the bindings menu in VR
2) Downloading them from the workshop
Since there currently are no bindings on the workshop I have been configuring these by hand, mostly copying the bindings from windows. This has gotten me some working bindings for SteamVR Home and the SteamVR Dashboard so far, SkyrimVR apparently counts as a legacy application, and for that I have yet to get the thumbsticks configured (judging from the windows config that should be axis0, but that doesn't exist as an option O.o). I'll try and publish the working configs for SteamVR Home and Dashboard to the workshop, in case someone would like to try them out :)

thaytan, 2020-09-02:
That sounds great! It should also be possible to find the bindings files steam has created in its directories somewhere and to put them directly into the upstream repo.

witty91, 2020-09-03:
Hm, I've been grepping through ~/.local/share/steam and  the only results are binaries, some workshop files and my personal bindings. Which matches what I found online - in order to export anything of this sort one has to create a personal binding which then goes into "~/.local/share/steam/userdata/$userid/$gameid". And also steam apparently really doesn't like one manually copying files into there :-D At least I wasn't able to load them afterwards.
The files in the directories mentioned above could of course go into the upstream repo.

witty91, 2020-09-14:
So I have been trying out mappings for skyrim vr the past week or so, and I got some mixed results with this mapping: [legacy_game.json.txt](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/5220436/legacy_game.json.txt)
But as far as I can tell the mapping is perfectly fine for rift s controllers, Skyrim VR just wrongly assumes them to be HTC Vive controllers, which is an issue on the game's side. I'm currently looking for another game to try this out with ;-)

I would consider this issue resolved, since it only depends on controller bindings within Steam VR now, everything else appears to work just fine :-)

thaytan, 2020-09-15:
Awesome! I'll look at incorporating that as the default legacy bindings on the weekend and see where it leads

thaytan, 2020-09-20:
If you get a chance, can you try the changes I just pushed? I incorporated your bindings - the legacy bindings and the dashboard ones from the steam workshop. I removed the references to non-existent touch reports for the joystick, trigger and grip.

Let me know if it's any better/worse. I had quite inconsistent results here - I need a more stable test machine for Linux/SteamVR.

witty91, 2020-09-20:
I just pulled it and tried it, seems to work just as expected for me! So, everything works in SteamVR Home and the Dashboard, and is still kinda broken in SkyrimVR because the game expects me to be holding HTC Vive controllers instead of Rift S ones. While looking for information on this I found a possible fix for this on the games' side, that I will have to try out still.

How this works for other headsets and controllers I don't know.

thaytan, 2020-09-20:
Thanks for the quick testing!

How does it break in SkyrimVR? Is it just that it's showing the wrong controller models, or is there something more going on?

I think fixing the models would be possible - but to really do it right, OpenHMD needs to provide model information for the SteamVR-OpenHMD wrapper to publish.

witty91, 2020-09-20:
The Mail reached me, when I was in front of my PC and had free time, therefore I could test it almost immediately.

The thumbsticks directions are not being recognized in SkyrimVR, when I use teleporting motion (like in SteamVR Home), it's bound to left stick click, which I didn't put there. I believe this is related to a similar issue with Index controllers:
https://www.nexusmods.com/fallout4/mods/41931
Which curiously enough gets fixed by making them act as oculus controllers on windows, apparently.
This has a short explanation to what this mod is doing:
https://forum.vive.com/topic/6649-no-support-for-cosmos-controllers/

But hey, I can teleport around and just burn everything down with magic :-D

witty91, 2020-10-04:
I just decided to get another legacy game to test on. In that game the thumbsticks are working perfectly fine, therefore I'm fairly certain that my previous issues with SkyrimVR are game specific and the bindings themselves are okay. Haven't solved those issues yet though...

thaytan, 2020-10-05:
I merged the controller branch back to master last night - that'll make it easier for more people to test