ChristophHaag/SteamVR-OpenHMD #33: Windows build only works with old subproject

l33tlinuxh4x0r, 2019-09-19:
If I try to compile a windows build using the latest git upstream for OpenHMD it fails. If I use the old version it builds. (Using cmake)

```
[ 83%] Linking C shared library libopenhmd.dll
/usr/lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/openhmd.dir/objects.a(openhmd.c.obj):openhmd.c:(.text+0xac0): undefined reference to `__imp_ohmd_list_open_device_s'
/usr/lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/openhmd.dir/objects.a(platform-win32.c.obj):platform-win32.c:(.text+0x133): undefined reference to `__imp_ohmd_sleep'
collect2: error: ld returned 1 exit status
make[2]: *** [subprojects/openhmd/CMakeFiles/openhmd.dir/build.make:462: subprojects/openhmd/libopenhmd.dll] Error 1
make[1]: *** [CMakeFiles/Makefile2:125: subprojects/openhmd/CMakeFiles/openhmd.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

ChristophHaag, 2019-09-19:
Is it this upstream issue? https://github.com/OpenHMD/OpenHMD/issues/245

l33tlinuxh4x0r, 2019-09-20:
yes

closed 2021-07-22 by l33tlinuxh4x0r