ChristophHaag/SteamVR-OpenHMD #35: How to setup Nolo and what I should expect from it

MagicRB, 2020-01-02:
I have been trying to get NoloVR working, so far no luck, I have figured out that if I connect via USB the thing that goes on my head, openhmd segfaults, so I might be doing something wrong inside my config file. I used the default one, for separate HMD and tracking. How did you get it working, I've seen a pic of it working somewhere.

ChristophHaag, 2020-01-03:
I would recommend first trying to get the HMD + nolo working with OpenHMD alone before trying it with SteamVR.

OpenHMD's `simple` example takes an enumerated device index as argument so if you get tracking data from all the different devices with `./simple 0`, `./simple 1`, etc. then you know at least that works.

The config file is a really rudimentary hack I only did because OpenHMD is lacking any configuration so far... If you run openhmd's simple example and immediately quit it with ctrl+c, you'll see a list of devices at the top, each enumerated with `device 0: ...`, `device 1`, ... (the numbers can even change when some additional VR hardware supported by openhmd is connected). I believe usually if you have a HMD and Nolo, the HMD should have 0, and the Nolo Tracker 1.

The last time I tried DK2 + Nolo was a long time ago. I don't see a reason why SteamVR-OpenHMD wouldn't work with it anymore, but I haven't tested it lately...

MagicRB, 2020-01-03:
I did manage to get it working even in SteamVR but the tracking was all borked, it did not track properly at all