ChristophHaag/SteamVR-OpenHMD #36: Can't get Steamvr work in Fedora with WMR Device

spiderbit, 2020-01-12:
I can my Samsung Odysee in extended mode, the tracking with the opengl example works, including when I start it as normal user.

But SteamVR always crashes and goes then to safe mode.

Error I see in Console is:

```
[II] 045e:0659 0002:0003:02

[II] opening

[EE] could not open device with index 0, check device permission?
```

herosilas12, 2020-05-22:
did you try with sudo?

spiderbit, 2020-05-23:
Well I tried other distros :D since then not only because of that but some packages where not available which were in AUR from Archlinux as example. I think I got it running, the problem was then that I was 90 degree sidewards aligned after the calibration and I have no way to press the buttons in the intro because the controllers are not supported.

But It got to the tracking part, but hard to pin down the exact steps steam user config seems to be a big mess where you change stuff and pray that the gods are with you.

Again could not finish the tutorial thing because there is no way to press a button.

spiderbit, 2021-02-17:
I switched to arch on that machine so that bug doesn't make sense anymore but I make a new one with the orientation problem.

closed 2021-02-17 by spiderbit