ChristophHaag/SteamVR-OpenHMD #37: How do you acquire linux64 builds?

gazhay, 2020-02-19:
Got as far as 

`~/.local/share/Steam/steamapps/common/SteamVR/bin/linux64/vrpathreg adddriver ~/SteamVR-OpenHMD/build`

But I do not have a linux64 directory, I have a linux32 directory.

Tried multiple steamVR betas but non create the linux64 drivers.

ChristophHaag, 2020-02-27:
Sorry no idea. If you find a `vrpathreg` executable anywhere in the SteamVR directory that should work, but I would be suprised if SteamVR itself works without the 64 bit directory. That's on a 64 bit distribution/kernel, right?

gazhay, 2020-02-27:
Well, that's what I thought - but Steam installed SteamVR (and any of the betas I try) and has never created the linux64 directory.

SteamVR opens and I can get system reports from it - which lists the Oculus Quest I have as a usb device, but doesn't recognise it as a headset.

The log shows that it tries to load oculus drivers from linux64 directory, but obviously fails.

Thanks for looking anyway.

TheOnlyJoey, 2020-06-15:
The Oculus Quest is currently not supported by OpenHMD so it makes sense it can not find it. Please check http://www.openhmd.net/index.php/devices/ for compatibility.