ChristophHaag/SteamVR-OpenHMD #40: issue during make (mingw32-make)

herosilas12, 2020-06-08:
[log file](https://drive.google.com/file/d/1ahFcO6sovUUFjLJvCN9blY5_LWjZNRSQ/view?usp=sharing)
IDK what to do here

thaytan, 2020-06-09:
Your log file isn't public. It'd be better to attach it directly here.

herosilas12, 2020-06-09:
[errors.txt](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/4752119/errors.txt)
sorry about that

thaytan, 2020-06-10:
You should be able to change the `#if defined(WIN32)` on the line above to `#if defined(_MSC_VER)` - the vsnprintf() variant should work on mingw32

herosilas12, 2020-06-15:
Now there are m_pWatchdogThread errors
[errors.txt](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/4782825/errors.txt)

thaytan, 2020-06-16:
I'm not sure what's going on there. AFAICS, the include <thread> is already there at line 9.
https://sourceforge.net/p/mingw-w64/discussion/723797/thread/67f35b9c/?limit=25 suggests that MingW might require an extra clause in the CMakeLists:

 `-std=gnu++11 compiler flag may also need to added in CMakeLists.txt like`

herosilas12, 2020-06-16:
where in cmakelists.txt would I put that?

thaytan, 2020-06-16:
Try like this, near the existing MSVC check:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index b176519..29e779f 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -26,6 +26,13 @@ else(MSVC)
   target_compile_options(driver_openhmd PRIVATE -Wall) #-Wextra -Wno-unused-parameter) #-pedantic)
 endif(MSVC)
 
+if( MINGW )
+  add_definitions(-D__STDC_FORMAT_MACROS)
+  add_definitions(-D__USE_MINGW_ANSI_STDIO=1)
+  set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=gnu99")
+  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=gnu++11")
+endif()
+
 target_link_libraries(driver_openhmd
   ${CMAKE_DL_LIBS}
   openhmd
```

herosilas12, 2020-06-16:
same issue :( 
[errors.txt](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/4787211/errors.txt)

herosilas12, 2020-06-16:
The following zip has a package list as well as a python tool for searching through it (it is REALLY long)
[packages.zip](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/4787333/packages.zip)

thaytan, 2020-06-17:
I'm a bit out of ideas at that point. I read something about there being 2 different threading models available to choose in MingW (win32 vs posix) that might make a difference, but I don't really know.

herosilas12, 2020-06-17:
yeah I saw that somewhere too and I did try posix but it also didn't work. :(
I am guessing you are on linux so you don't have to deal with this kinda stuff?
do you think cygwin would work?

thaytan, 2020-06-17:
I'm on Linux, yes. Although if I was going to try compiling on Windows these days, I'd first try the Meson build with MSVC before MingW

herosilas12, 2020-06-17:
I tried meson and ninja first
unfortunately ninja also spits out m_pwatchdogthread errors
I don't have enough uses for visual studio to install it for one project
VScode works fine for what I do (I work in python on small tools that I use)

thaytan, 2020-06-17:
I understand. For Windows at least, it would be great if we could provide releases here - but this project isn't quite advanced enough for that yet

herosilas12, 2020-06-17:
I tried use msvc 2019 after running meson --backend vs
and here is the log after using "msbuild steamvr-openhmd.sln"
[errors.txt](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/4792737/errors.txt)

ChristophHaag, 2020-06-17:
About mingw: I don't know windows well but there is a possibility that SteamVR can't even load plugins built with mingw. As Valve's OpenVR driver header clearly doesn't support building with mingw, I would too recommend to try with msvc first.

CMake also has generators for various msvc/visual studio versions: https://cmake.org/cmake/help/v3.14/manual/cmake-generators.7.html#visual-studio-generators but they won't help with the errors you're seeing now.

    driver_openhmd.cpp(51,13): error C2131: expression did not evaluate to a constant

I remember reading that msvc does not support variable length arrays. Changing this in the code should be pretty simple.

    driver_openhmd.cpp(555,48): error C2065: 'M_PI': undeclared identifier

yea I guess M_PI is not in a C standard. Easy fix should be

```c
#ifndef M_PI
#define M_PI (3.14159265358979323846)
#endif
```

    driver_openhmd.cpp(649,22): error C2513: 'float': no variable declared before '='

really, msvc? https://stackoverflow.com/questions/118774/is-there-a-clean-way-to-prevent-windows-h-from-creating-a-near-far-macro

    driver_openhmd.cpp(665,1): error C2059: syntax error: ','

`DriverLog()` may be a different function prototype on windows?

TheOnlyJoey, 2020-06-17:
Actually, i think other then having to add the M_PI line, i got this compiled with meson and the provided build-win64.txt without any issues on linux.... So don't really know.
I did have to do the near/far macro thing (i think i just undef'ed it).
Worked fine on a Windows 10 setup.

herosilas12, 2020-06-17:
how do I use build-win64.txt
meson ./configure?
can I set it in cmake-gui?

TheOnlyJoey, 2020-06-17:
Meson is a different build system from cmake.
I only used it to build Windows binaries from Linux (https://mesonbuild.com/Cross-compilation.html)

herosilas12, 2020-06-17:
I fixed the most of the stuff using what you suggested except for the "expression did not evaluate to a constant" error
[errors.txt](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/4793481/errors.txt)

herosilas12, 2020-06-17:
I download mingw-w64 and specified the 64bit compilers (they have posix I think)
and it built fine but when I tried to install it says it can't find "openhmd.pc"