ChristophHaag/SteamVR-OpenHMD #42: Rift S Controllers: Thumbsticks only registered in the first quadrant

witty91, 2020-08-25:
Hi, when using the Rift S Controllers in SteamVR Home, I noticed that the thumbsticks only register an input, when it is within the first quadrant (so, positive x and y values). I have been looking through the code but haven't found anything. So I would like to ask two questions:
1) Does anyone else have this problem? (It does work in windows, so the hardware should be fine...)
2) Can anyone else find a bug in the code that I couldn't?

thaytan, 2020-08-26:
2nd time I've seen something similar reported. I just double checked, and OpenHMD is outputting the full range of the stick, but it does it over -1.0 to 1.0... perhaps SteamVR is expecting 0.5 as the rest position, and 0..1.0 as the output range

witty91, 2020-08-26:
That might actually have been me on another thread...since I couldn't figure the issue out until now, I thought I should create a separate issue. 
From what I found online -1 to 1 should be the correct range. I'll try to find the link to this information again.

thaytan, 2020-08-26:
Check using OpenHMD's simple example that you're getting the same range I am:

`./build/openhmd_simple_example 1`

and then move the left controller stick and watch the controls state line for example.

thaytan, 2020-08-26:
Ah, I see the problem - all the analog controls are reported to SteamVR as VRScalarUnits_NormalizedOneSided, but that should only apply to trigger and grip controls. The joystick report is VRScalarUnits_NormalizedTwoSided

witty91, 2020-08-26:
Yes, that fixes it! For now I have just changed "VRScalarUnits_NormalizedOneSided" line 300 in driver_openhmd.cpp to "*TwoSided" and don't see any adverse effects. I reckon another if condition to catch the thumbsticks would be better, but I am not sure by what property I could get them.

Anyways, this is working really well! I finally can turn left :-D

thaytan, 2020-08-26:
Thanks! I'll make a patch

thaytan, 2020-08-27:
Please update and test! :)

witty91, 2020-08-27:
I just had some time to try it out, and it works perfectly fine! :) 
Now I only need to map the buttons correctly for Skyrim VR and one could at least play as a caster, where you don't need to wave a sword around :D

thaytan, 2020-08-27:
If you know how input mappings work... feel free to email me and explain :)

witty91, 2020-08-28:
I have managed to recreate the SteamVR Home bindings from windows, so I can use the tools to draw and so on. The way I do this is:
1) Find the bindings on windows
2) In SteamVR Home open the menu and configure the buttons through their interface
3) Use an xbox360 controller if I screw up the mapping inbetween
Since I'm not at my Desktop PC for the weekend, I'll try to create a mapping for SkyrimVR next week. I could create a more detailed guide of that process if that's helpful :)

commit 3c581f61 (2020-08-27): Mark analog stick controls as two-sided reports

Make SteamVR aware that analog stick controls range from
-1.0 to 1.0 by marking them as two-sided. Otherwise only
the top-right quadrant of the stick range is usable.

Fixes #42
commit 00a2b370 (2020-10-04): Mark analog stick controls as two-sided reports

Make SteamVR aware that analog stick controls range from
-1.0 to 1.0 by marking them as two-sided. Otherwise only
the top-right quadrant of the stick range is usable.

Fixes #42
closed 2020-09-01 by witty91