ChristophHaag/SteamVR-OpenHMD #43: [AMD | VR-Tek WVR2] vrcompositor searches for resolution at incorrect orientation resulting in Direct Mode failure

Sudospective, 2020-08-31:
VR-Tek being the funky wacky HMD it is has an internal resolution of 1440x2560, that being width by height. When SteamVR is started, vrcompositor attempts to search for the same resolution, but inverted (i.e., 2560x1440) and results in being "unable to find a Direct Mode compatible resolution". Currently using the `master` branch of SteamVR-OpenHMD and the `master` branch of OpenHMD. Here's a snippet of the log:
```
Mon Aug 31 2020 08:52:57.000914 - HMD deviceUUID is 7c4734bb84d31201
Mon Aug 31 2020 08:52:57.000959 - Looking for direct display through RandR
Mon Aug 31 2020 08:52:57.000978 -  - Root 0x5b1
Mon Aug 31 2020 08:52:57.419875 -    - Output 0x42 - 67 modes, 1 preferred
Mon Aug 31 2020 08:52:57.420012 -      - Mode 0 0x4a 1366x768 (looking for 2560x1440)
Mon Aug 31 2020 08:52:57.420529 -    - Output 0x43 - 0 modes, 0 preferred
Mon Aug 31 2020 08:52:57.420957 -    - Output 0x44 - 2 modes, 1 preferred
Mon Aug 31 2020 08:52:57.421071 -      - Mode 0 0x8d 1440x2560 (looking for 2560x1440)
Mon Aug 31 2020 08:52:57.421219 -    - Output 0x45 - 0 modes, 0 preferred
Mon Aug 31 2020 08:52:57.421889 -    - Output 0x46 - 0 modes, 0 preferred
Mon Aug 31 2020 08:52:57.422062 -    - Output 0x47 - 0 modes, 0 preferred
Mon Aug 31 2020 08:52:57.422204 -    - Output 0x48 - 0 modes, 0 preferred
Mon Aug 31 2020 08:52:57.422273 - Tried to find direct display through RandR: (nil)
Mon Aug 31 2020 08:52:57.422321 - Looking for direct display through Vulkan WSI
Mon Aug 31 2020 08:52:57.422371 - Tried to find direct display through Vulkan WSI: (nil)
Mon Aug 31 2020 08:52:57.422414 - CHmdWindowSDL: Failed to create direct mode surface
```
The thing I'm not sure about is if it's an issue with SteamVR, OpenHMD, or this here bridge for the two. Sorry in advance if this is in the wrong repository.

Sudospective, 2020-08-31:
I just opened up the `vrtek.c` file from the OpenHMD source, and they have a commented line that states that the user is expected to have rotated their HMD display by using the command `xrandr --output HDMI-A-1 --rotate left`. It seems pretty unfair to only include that in commented code in an obscure C-file that a non-developer would never dare touch, but that's what they propose. Might be good to put that in the docs if it isn't in there already.

ChristophHaag, 2020-08-31:
So openhmd does not provide display rotation information at the moment. The projection matrices it provides do transparently rotate the images appropriately (I believe for the WVR2 this is the case but not 100% sure right now).

SteamVR-OpenHMD has some dodgy code to extract the rotation from the projection matrix, and rotates the images in the distortion code https://github.com/ChristophHaag/SteamVR-OpenHMD/blob/f3840280d99208bc58edd6c55cd134d2e199b3d4/driver_openhmd.cpp#L676-L690.

Now the issue is that we have to tell SteamVR about the rotated orientation of the display so it searches for the right resolution. I am pretty sure this is possible, for example the DK2 had a rotated display too. If nowhere else, the solution may be found in OSVR-Oculus-Rift + SteamVR-OSVR.

Once we figure out how to tell SteamVR about the display rotation we may have to remove the rotation hack from the distortion function because it's possible that SteamVR can handle it internally.

But we should make sure it also works with more exotic rotations like the LG 360 which I believe has the left display rotated 90° counter clockwise and the right display 90° clockwise.

Sudospective, 2020-08-31:
I used to have the Windows drivers for this VR-Tek headset. I wonder if it does it through the VR settings JSON...

Sudospective, 2020-08-31:
https://developer.valvesoftware.com/wiki/SteamVR/steamvr.vrsettings Valve has entries for driver sections in here that allow to specify width and height, but I'm not sure about rotation. However, I have noticed that when the output is manually rotated clockwise, it seems to have the two halves of the screen rotated counter-clockwise in each respective eye. It's working with a really cursed matrix for sure.

Sudospective, 2020-09-01:
As expected, rotating the HMD display allows it to display to the HMD. However, SteamVR still reports that Direct Mode is disabled. It's possible I missed a step during setup, but I can't quite make out why this is occurring. I'm gonna take a more thorough look through the logs and attach anything that might shed some light.

JSaurusRex, 2020-10-03:
https://github.com/ValveSoftware/SteamVR-for-Linux#known-issues

> The VR status window isn't currently aware of direct mode being enabled or not, so the "enable direct mode" and "disable direct mode" buttons should not be used; direct mode is automatically enabled where supported