ChristophHaag/SteamVR-OpenHMD #46: SteamVR not detecting CV1 with thaytan's latest openHMD rift positional tracking branch

hanaral, 2020-12-04:
Despite the latest additions to make it work(sometimes) SteamVR does not seem to detect the headset whatsoever. I've checked the vrserver log and it seems to complain that the build dir (that is used as the external driver) is not a directory. 
The OpenHMD side of things seem to work, the work of @thaytan seems to be showing in the openGL test, so I presume its on the SteamVR end. 
Please tell me if there are any other info I can give to help!

hanaral, 2020-12-04:
Seems to work with cmake

closed 2020-12-04 by hanaral

thaytan, 2020-12-05:
If you want to run against my dev branch of OpenHMD, you need to go into the `subprojects/openhmd` subdir of SteamVR-OpenHMD and manually point it at the right version of OpenHMD

hanaral, 2020-12-05:
@thaytan That’s how I was doing it, not to matter since it seems to work fine with cmake