ChristophHaag/SteamVR-OpenHMD #53: Oculus rift s tracking working but not displaying

slop-me, 2021-01-25:
I got the tracking of the headset going but the Oculus rift s headset but the headset won't display anything or even show up in xrandr almost like the Oculus rift s is sleeping. Any suggestions on what to try

thaytan, 2021-01-25:
This might be related to https://github.com/OpenHMD/OpenHMD/issues/305

Also, just in case - the OpenHMD Rift S driver doesn't turn on the screen unless the prox sensor is triggered, so you need to actually put it on or cover the sensor with your thumb.

thaytan, 2021-01-25:
Also check `xrandr --prop` output. If the display is present, it may show up as a non-desktop display and therefore not be in the default `xrandr` output.

slop-me, 2021-01-25:
I put it on and it still doesn't show up in xrandr I thought that might be an issue. Do you still need to do xrandr --prop even with it on?

thaytan, 2021-01-25:
I wasn't clear - the headset is alive and creates a display connection as soon as SteamVR is running, but the *screen* is left switched off until the prox sensor is triggered.

Even if you're not wearing the headset, it should show up as a non-desktop display in `xrandr --prop`

slop-me, 2021-01-25:
So how do I get the steamvr window off my desktop and onto the display sorry you have me rather confused now.

ChristophHaag, 2021-01-25:
Currently if the HMD has a portrait mode display that is physically built in in landscape mode, it can only be used with extended mode in SteamVR-OpenHMD https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode

GetWindowBounds currently returns the display dimensions of the display "after" rotation, as the user sees it. It needs to return the display dimension "before" rotation, like the physical layout that the operating system sees at the display connector / in xrandr.

GetEyeOutputViewport needs to set up where the left and right eyes should be shown on the display, this should be coordinates on the "before" rotation configuration too.

The distortion function already takes care of rotating what is actually rendered and there is some hacky code to extract the rotation from the openhmd projection matrix in there. Maybe we should just hardcode which product names need display rotation.

ChristophHaag, 2021-01-25:
In the Monado SteamVR driver that works already https://gitlab.freedesktop.org/monado/monado/-/blob/master/src/xrt/state_trackers/steamvr_drv/ovrd_driver.cpp

That implementation is simple because the drivers do the work to set it up right, see comments for xrt_view and xrt_hmd_parts. https://gitlab.freedesktop.org/monado/monado/-/blob/master/src/xrt/include/xrt/xrt_device.h

The Monado OpenHMD drvier hardcodes a bunch of displays with their rotations https://gitlab.freedesktop.org/monado/monado/-/blob/master/src/xrt/drivers/ohmd/oh_device.c