ChristophHaag/SteamVR-OpenHMD #54: SteamVR-OpenHMD fails to compile.

hedmo, 2021-01-29:
hi . i am trying to compile SteamVR-OpenHMD but it fails with : 

CMake Error at subprojects/openhmd/cmake_install.cmake:95 (file):
  file INSTALL cannot find
  "/var/tmp/portage/media-libs/SteamVR-OpenHMD-20201004/work/SteamVR-OpenHMD-20201004_build/openhmd.pc":
  No such file or directory.
Call Stack (most recent call first):
  cmake_install.cmake:47 (include)

FAILED: CMakeFiles/install.util 
cd /var/tmp/portage/media-libs/SteamVR-OpenHMD-20201004/work/SteamVR-OpenHMD-20201004_build && /usr/bin/cmake -P cmake_install.cmake

regards hedmo
[build.log](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/5896125/build.log)

TheOnlyJoey, 2021-02-03:
SteamVR-OpenHMD uses a submodule for openhmd so if setup correctly should automatically build. There is some information in the Readme about that.
Also love the avatar MM7 is my favorite ;)

hedmo, 2021-02-03:
@TheOnlyJoey . as far a i can see in the build.log  openhmd is compiled but fails because ti cannot find openhmd.pc. correct me if i am wrong.

thaytan, 2021-02-03:
I don't know that anyone has tested actually running `make install` for SteamVR-OpenHMD. Usually, one just builds it and runs `./register.sh` to register the plugin files directly from the build dir.

We have discussed allowing using an external OpenHMD before - it is a good idea, just noone has done the build changes for that yet

hedmo, 2021-02-03:
@thaytan thank you for the reply. as i am running all my "custom" package (not provided by gentoos package manager ) it is better to have them tracked there 
either way to not have conflicts in the future and clean out deps afterwards.i hope this can be fixed so i can put this global over all of my boxes as i have 4 vr sets.and using my github for manage my custom packages.