ChristophHaag/SteamVR-OpenHMD #57: No Steam games launches in VR on Linux

LeGrandToto, 2021-02-25:
On a up-to-date Manjaro install using a gnome session, I have installed xrhardware, compiled OpenHMD, compiled this project, installed SteamVR (up-to-date on its current beta "beta - SteamVR Beta Update"), registered the driver using the script and now SteamVR successfully detects and use the PSVR I have connected to my computer.
But no game seems to be able to run in VR, they launch on my main monitor instead (in 2D with no headtracking) regardless if launched from SteamVR or from Steam (and selecting "Launch in Steam VR mode"). SteamVR will show the (for instance) "next up... Elite Dangerous" prompt but the game just launches in 2D on my main monitor and SteamVR stays in its space background.

Here are other informations you might find useful:

Hardware:
- CPU: Threadripper 3960X
- GPUs:
  - NVIDIA GeForce RTX 2080 SUPER (it has my monitor and the PSVR attached to it)
  - NVIDIA GeForce GTX 960 (Nothing is connected to it, `nvidia-smi` tells me Xorg also runs on it though)

Other tests:
- I have also tried to start a VR session in Blender 2.91.2 using `XR_RUNTIME_JSON=~/.steam/steam/steamapps/common/SteamVR/steamxr_linux64.json blender` for blender to find the OpenXR session, SteamVR shows the "next up... blender" prompt but then blender crashes on a Segmentation fault...
- I have also installed on its own partition Ubuntu 20.04 and repeated the installations made on Manjaro with the all the same results.
- I have also tried to use the PSVR on Windows 10 but I have no clue how to compile and install the driver. I think I managed to compile this project correctly and register it and my last problem is to give Windows/Steam "the USB driver for the PSVR" (or what xrhardware does for the linux world, I am not a hardware/driver guy).
- I have successfully run Elite Dangerous in VR and the VR viewer of Blender using the PSVR on Windows using the free demo for Trinus VR...
- I have changed the values used in OpenHMD for the PSVR to make sure OpenHMD was used and yeah, I can add a barrel distortion effect in SteamVR by adding one in `SteamVR-OpenHMD/subprojects/openhmd/src/drv_psvr/psvr.c`

Last note:
I have once had Elite Dangerous run in VR pretty much flawlessly for an hour on Manjaro. I was changing the command line options for SteamVR to `OHMD_VENDOR_OVERRIDE=Oculus %command%` or `OHMD_VENDOR_OVERRIDE=HTC %command%` and flipping the `non-desktop` flag for the PSVR at random. I tried multiple times to find what I did right but I haven't managed to have that game open correctly again!

ChristophHaag, 2021-02-26:
OpenXR + OpenGL apps like blender not working may be
https://steamcommunity.com/app/250820/discussions/8/3113647550048444866/
https://github.com/ValveSoftware/SteamVR-for-Linux/issues/421
That only affects OpenXR + OpenGL apps, not OpenVR though.

> Windows 10 but I have no clue how to compile and install the driver

Welcome to the club. You probably need some development experience to compile it on windows. I'm told it can be done and that some people want to put something together to make it easier on windows.

> OHMD_VENDOR_OVERRIDE=Oculus %command%

Afaik the only thing that uses this info is SteamVR Home, and it enables a gaze pointer if that is set to Oculus.

> and flipping the non-desktop flag for the PSVR at random

doesn't do anything on nvidia, only amd and intel use it.

If the steamvr compositor displays and tracks properly, then the work of SteamVR-OpenHMD "is done" and the rest of handling games is up to SteamVR itself and I would look for the problem somewhere there.

Some help may be the log file ~/.steam/steam/logs/vrserver.txt. SteamVR also creates a log file in that directory for every individual game.

LeGrandToto, 2021-02-26:
Thank you so much for replying so quickly! I will mainly head to the SteamVR team then for my problem of games not launching/attaching to the compositor.
I might give it an other try at making this project run on Windows and maybe post somewhere here the steps I followed but that might take me a month if it happens at all.

A few last notes:

I forgot to mention: SteamVR complains the headset is running as a monitor and I never manage to have the headset in direct mode instead.

> Afaik the only thing that uses this info is SteamVR Home, and it enables a gaze pointer if that is set to Oculus.

When in SteamVR, I use a plugged PS4 controller to enter any kind of inputs, I can hold the SHARE button and it opens a menu with my recently played VR games and an action bar at the bottom (where I can show the desktop, exit SteamVR Home, exit SteamVR...). In that menu I can select things by looking at them (I have a blue dot in front of me at all times). This "gaze pointing" is available to me no matter if or what value I entered for `OHMD_VENDOR_OVERRIDE`. If "gaze pointing" is meant to describe something else, I have never noticed what it is. I for instance never managed to walk or grab anything while in the house. I can press R2 and it shows me where I would teleport but I never moved to that location.

> doesn't do anything on nvidia, only amd and intel use it.

for disambiguation: Changing that value does change things for my desktop:
- When `non-desktop` is `0` (the default value): `xrandr` lists the PSVR as `connected` and I can use the TV connected to the Linkbox as a secondary monitor with no problems (when the box is not in its VR mode, otherwise, it only shows me an undistorted crop of the right eye screen). SteamVR works correctly and displays correctly in the PSVR. After using SteamVR I need to restart the PSVR if I want to correctly use the TV as a monitor again (otherwise, it remains stuck in its VR mode and displays only a fraction of the desktop).
- When `non-desktop` is `1`: `xrandr` lists the PSVR as `disconnected`, gnome's display settings only lists my main monitor BUT I can still put my mouse pointer where the PSVR desktop used to be and the pointer will be displayed correctly, I can still put a window there and the window will be displayed correctly but the desktop/wallpaper will display artefacts of the window. It is as if gnome is not longer managing that screen anymore but the X11 buffer was still allocated and the PSVR displays that buffer... Somehow... And again, SteamVR works correctly and displays correctly in the PSVR.
- When I set gnome to only use my main monitor, SteamVR launches on the main monitor with correct head tracking and the PSVR (and the TV attached to it) says it is not connected to an HDMI source.
- I have read (although not fully understood) [this page](https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode) and created `/usr/share/X11/xorg.conf.d/99-HMD.conf` with:
```
Section "Device"
    Identifier     "Device0"
    Driver         "nvidia"

    Option "AllowHMD" "yes"
EndSection
```

I am happy to close this issue as it mainly affects other libraries but it might also be helpful for people to find... I have never created an issue on a public repo so I don't know what the best practices are...

LeGrandToto, 2021-02-26:
Quick update: I have now realised that The Forest launches correctly (it used to not work well.... and the performances in desktop mode are unacceptable in my case: in frame every 3 seconds or so). Now I can't play anything as I think I need to have a controller that allows me to point at it and select "start" but that is enough for me to start messing with things!
I have started by pairing my PS moves to my PC using [this](https://psmoveapi.readthedocs.io/en/latest/index.html), I also merged the `psmove` branch of openhmd to `master`, compiled, installed, compiled this project and re-registered the driver and one of the PS move light up red when I start SteamVR! (SteamVR also gives me a 307 error but that is my next task).

I also created a [bug report](https://github.com/ValveSoftware/SteamVR-for-Linux/issues/431) as I said. I think I can close this issue now but as I said in my last message, I do not know what the best practices are.

ChristophHaag, 2021-02-28:
> I forgot to mention: SteamVR complains the headset is running as a monitor and I never manage to have the headset in direct mode instead.

It's not very well known how SteamVR finds display outputs to use for direct mode. As far as I know the only requirement on nvidia should be that the display is currently not enabled/in use for extended desktop on X.org. ~/.steam/steam/logs/vrcompositor.txt may or may not give a hint about direct mode.

> When non-desktop is 1: xrandr lists the PSVR as disconnected

Interesting, if the nvidia driver really does that, that is new behavior.

However 

> gnome's display settings only lists my main monitor BUT I can still put my mouse pointer where the PSVR desktop used to be and the pointer will be displayed correctly, I can still put a window there [...]

sounds like buggy behavior. When the display output goes into "disconnected" state, the desktop is supposed to stop using that display for extending the desktop. Basically it's supposed to do  `xrandr --output HDMI-A-0 --off` (or whatever the output name with the HMD is).

> I have read (although not fully understood) this page and created /usr/share/X11/xorg.conf.d/99-HMD.conf

tl;dr: You need this config file and extended mode when you want to run one of the (very few) OpenHMD applications that use OpenHMD directly, without SteamVR. You do not need it if you want to use SteamVR though my current understanding is that having this config file doesn't prevent direct mode from working, as long as the display output is not in use by X.org.

Maybe this helps, maybe not? https://monado.freedesktop.org/direct-mode.html

For the controllers I don't think anyone has tested the psmove controller branch with SteamVR-OpenHMD and I'd say it's 50:50 if that actually works in any way.

You can also check out Monado's SteamVR driver. That won't solve any of your SteamVR issues, but the psmove controllers should work there: https://monado.freedesktop.org/steamvr.html

We also have basic positional tracking there, if you have the ps4 camera with the usb adapter. https://monado.freedesktop.org/positional-tracking-psmove.html

If you have the camera but not the adaptert, sony will send you one if you have a valid psvr serial number btw: https://camera-adaptor.support.playstation.com/

LeGrandToto, 2021-02-28:
> You can also check out Monado's SteamVR driver.

I was planning to. I had had a look at Monado when initially experimenting in connecting the PSVR to anything running on my computer. I had it compiled correctly (I suppose) but never managed to see anything correctly reacting to the PSVR... I wanted to blame it on the lack of cameras so I indeed ask Sony for the USB adaptor and it arrived yesterday. I will ultimately take inspiration from their work to add support for the PS moves in this project if I take the time to make Monado work.

But from my (limited) understanding, Monado is an open implementation of OpenXR that is a/the new standard to do all VR/AR/MR, with any hardware and any software that implements their part of the API. I *assume* that unfortunately the vast majority of early VR games do not implement this new API and instead use either Valve's OpenVR or Oculus' API that I never know the name of. I am still looking for a list or some kind of test to know if a game tries to launch in VR and what API it uses to do so. 

My personal ambitions in joining this project was to spend some time during this pandemic to make my PSVR work on steam games I own... And making whatever progress I may make open so others can join or build up on my work.

closed 2021-03-14 by LeGrandToto