ChristophHaag/SteamVR-OpenHMD #59: [Pavlov VR] {Linux} Controller tracking disconnects

Doomsdayrs, 2021-03-17:
Upon entering the game "Pavlov VR", Controller tracking cuts out. 
This is through https://github.com/thaytan/OpenHMD/tree/rift-kalman-filter

The controllers work completely fine (outside of lacking proper key binding) in the SteamVR Home, but upon entering Pavlov it immediately cuts out. I can see below me the line for a controller being stuck on the ground

Doomsdayrs, 2021-03-17:
If logs are needed, I can get them

thaytan, 2021-03-17:
Does that meant the controllers stop working permanently, or just while you're in that game and they come back if you exit?

The first one... I have no idea what could cause that. The 2nd might be something about the controller binding setup - maybe Pavlov VR doesn't use legacy bindings or something

Doomsdayrs, 2021-03-17:
Its the second one
Buttons also don't work, so one cannot simply press menu to get out, its just stuck

Doomsdayrs, 2021-03-17:
Using "Show Old Binding UI" allows access to the bindings, "No bindings found"

Let me try to create a new one

Doomsdayrs, 2021-03-17:
This is the default Pavlov Binding for the Oculus Touch
[Pavlov VR Default Oculus Touch Binding.txt](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/6158864/Pavlov.VR.Default.Oculus.Touch.Binding.txt)

It seems this game also requires bindings for Pose, Haptics, Skeleton

Maybe the issue is that as I can see, we have no output for Skeleton? Below is the bindings I have made, But I can't add skeletons

[image]

Doomsdayrs, 2021-03-17:
Here is the binding file I created. I also shared it through the "Publish to Workshop"
[[OpenHMD] {Pavlov VR} Oculus Touch.txt](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/6158947/OpenHMD.Pavlov.VR.Oculus.Touch.txt)

thaytan, 2021-03-20:
I don't have that game to try. How does it behave with those bindings?

Doomsdayrs, 2021-03-22:
I recently gotten the an RX6700xt
Currently drivers are bugged, and only today amdgpu microcode has been added to the kernel. So I have to wait for an update

Doomsdayrs, 2021-03-30:
Using the new controller and latest commit it works. I was having an issue with one of the stands not lighting up and my audio not working properly.

Now on Ubuntu 21.04, Using proper AMDGPU kernel drivers

Doomsdayrs, 2021-03-30:
Audio is due to weird ubuntu issues, can be ignored

thaytan, 2021-03-30:
Thanks for the update!

Doomsdayrs, 2021-03-30:
Alright after playing for a bit
1. I have motion sickness
2. I learnt a few shortcomings and flaws 

Because the game focuses heavily on stable shots and accurate mappings of where controllers are, OpenHMD fails to meet up. 

Ingame:

1. Controllers would swap places often when "holding" a gun with both hands. Likely confused which controller is which when they are put behind each other
2. Controllers have a constant jitter / juttter that makes holding a weapon in game still difficult 
3. Graphical issues, not sure if its because of improper specification of what resolution the head set is or the bleeding edge drivers of my new gpu. 

some notes

A solution to unswap hands ingame is to put your controllers together, and spin them around. "Untangling" them
Do not use 2 handed weapons. A pistol and knife are more stable then 2 hands trying to hold something

I recommend you @thaytan  to get the game. It has offline missions and such where you can setup and experiment with gameplay that requires more fine movement. Allowing you to have insight in where you can improve :D

thaytan, 2021-03-30:
That sounds like a fair assessment of the current state of things.

I think the graphical issues are related to SteamVR reprojection. See https://github.com/ValveSoftware/SteamVR-for-Linux/issues/227 and https://github.com/ValveSoftware/SteamVR-for-Linux/issues/277 for example

Doomsdayrs, 2021-03-30:
You used the same link for both :p