ChristophHaag/SteamVR-OpenHMD #60: WVR2 1440x2560 is actually 1440x24560

dtruesdale88, 2021-03-17:
I have played with rotation, resolution, scaling and tried every possible combination to get a clear screen in both eyes of the WVR2. I have had zero success. The closest I have been able to get is setting the scaling to match the resolution of 1920x1080. For some reason if I do not rotate the screen left in nvidia config, then everything is running vertical instead of horizontal. I have tried viewport in setting and viewport out settings and it keeps forcing viewport out to 1440x2560. Viewport in is 2560x1440 as long as the screen is rotated. SteamVR detects the hmd and it uses Openhmd, so I am trying to figure out what is wrong as to why I can not get this to display correctly.

ChristophHaag, 2021-03-17:
Also relevant for other HMDs with rotated displays: #53

I gave it a shot with this PR: #58. I think my WVR2 displays correctly in direct mode with this one but there might still be something wrong with it, especially the SteamVR preview window looks wrong.

fluffrabbit, 2021-03-17:
There are so many changes now that I can't tell if you fixed WVR2 rotations or not. I can test, but I know there are other issues with AMD cards.

fluffrabbit, 2021-03-17:
Tested with the `direct_mode_rotation` branch. Rotations are correct, but there is gyro drift and graphical glitches. Graphical glitches are worst on AMD, but Intel users shouldn't see serious problems.

dtruesdale88, 2021-03-18:
Ok tested out and in SteamVR is great, not sure if Direct mode is actually working. Launch game (Elite Dangerous) and it goes back to the rotation issue. Also it is treating is as a monitor, not HMD. have a I done something. Like adding Option AllowHMD yes to the nvidia config?

fluffrabbit, 2021-03-18:
3 things I'd like to say on the viability of WVR2 with OpenHMD:

1. It's on a list of devices that have buggy video output with AMD on Linux, along with PSVR and OSVR HDK2.

2. The drift may be mitigated by a Kalman filter that Joey is planning to port to the WVR2 driver and other headsets.

3. The drift varies with the temperature of the headset, and initial gyro offset values OpenHMD currently uses aren't necessarily correct.

dtruesdale88, 2021-03-19:
Agreed, is the AMD issue an AMD driver issue that needs to be pointed out to Red team?

I find the HMD very usable once the Drift issue is fixed in some manner. If this filter get's ported into the driver what will it do or fix? I love the work both projects have done. I have a HMD that actually works thanks to them, just needs a tweak.

fluffrabbit, 2021-03-19:
>Agreed, is the AMD issue an AMD driver issue that needs to be pointed out to Red team?

It's the open source Mesa driver. You know how these things go. Corps contribute code, but priorities are shit.

>If this filter get's ported into the driver what will it do or fix?

Sensor fusion between linear movement and rotational movement, which is supposed to smooth out the experience in a pleasing way, though ymmv. We'll see.

dtruesdale88, 2021-03-19:
Since your using the Open driver have you tried the closed to see if it works since it now works about 80% for gaming?

fluffrabbit, 2021-03-19:
No, Mesa drivers for Intel and AMD are supposed to be the best on Linux. Nvidia is the odd one out there. Anyways, I'm done with "gaming".