ChristophHaag/SteamVR-OpenHMD #62: error on cmake. ..

Bozdoc84, 2021-04-27:
cmake ..
CMake Error at CMakeLists.txt:13 (add_subdirectory):
  The source directory

    /home/bozdoc84/SteamVR-OpenHMD/subprojects/openhmd

  does not contain a CMakeLists.txt file.

SteamVR plugin path: /home/bozdoc84/SteamVR-OpenHMD/build/bin/linux64
-- Configuring incomplete, errors occurred!
See also "/home/bozdoc84/SteamVR-OpenHMD/build/CMakeFiles/CMakeOutput.log".

..............................................................HERE IS CMakeOutput.log   THANKS.................................................................................
                                                               

The system is: Linux - 5.6.17-050617-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/bozdoc84/SteamVR-OpenHMD/build/CMakeFiles/3.10.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/bozdoc84/SteamVR-OpenHMD/build/CMakeFiles/3.10.2/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/bozdoc84/SteamVR-OpenHMD/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_8da43/fast"
/usr/bin/make -f CMakeFiles/cmTC_8da43.dir/build.make CMakeFiles/cmTC_8da43.dir/build
make[1]: Entering directory '/home/bozdoc84/SteamVR-OpenHMD/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_8da43.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_8da43.dir/testCCompiler.c.o   -c /home/bozdoc84/SteamVR-OpenHMD/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_8da43
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_8da43.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_8da43.dir/testCCompiler.c.o  -o cmTC_8da43 
make[1]: Leaving directory '/home/bozdoc84/SteamVR-OpenHMD/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/bozdoc84/SteamVR-OpenHMD/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_1d37a/fast"
/usr/bin/make -f CMakeFiles/cmTC_1d37a.dir/build.make CMakeFiles/cmTC_1d37a.dir/build
make[1]: Entering directory '/home/bozdoc84/SteamVR-OpenHMD/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_1d37a.dir/CMakeCCompilerABI.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_1d37a.dir/CMakeCCompilerABI.c.o   -c /usr/share/cmake-3.10/Modules/CMakeCCompilerABI.c
Linking C executable cmTC_1d37a
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_1d37a.dir/link.txt --verbose=1
/usr/bin/cc     -v -rdynamic CMakeFiles/cmTC_1d37a.dir/CMakeCCompilerABI.c.o  -o cmTC_1d37a 
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
[... 529 lines cut ...]
    Feature record: CXX_FEATURE:0cxx_long_long_type
    Feature record: CXX_FEATURE:0cxx_noexcept
    Feature record: CXX_FEATURE:0cxx_nonstatic_member_init
    Feature record: CXX_FEATURE:0cxx_nullptr
    Feature record: CXX_FEATURE:0cxx_override
    Feature record: CXX_FEATURE:0cxx_range_for
    Feature record: CXX_FEATURE:0cxx_raw_string_literals
    Feature record: CXX_FEATURE:0cxx_reference_qualified_functions
    Feature record: CXX_FEATURE:0cxx_relaxed_constexpr
    Feature record: CXX_FEATURE:0cxx_return_type_deduction
    Feature record: CXX_FEATURE:0cxx_right_angle_brackets
    Feature record: CXX_FEATURE:0cxx_rvalue_references
    Feature record: CXX_FEATURE:0cxx_sizeof_member
    Feature record: CXX_FEATURE:0cxx_static_assert
    Feature record: CXX_FEATURE:0cxx_strong_enums
    Feature record: CXX_FEATURE:1cxx_template_template_parameters
    Feature record: CXX_FEATURE:0cxx_thread_local
    Feature record: CXX_FEATURE:0cxx_trailing_return_types
    Feature record: CXX_FEATURE:0cxx_unicode_literals
    Feature record: CXX_FEATURE:0cxx_uniform_initialization
    Feature record: CXX_FEATURE:0cxx_unrestricted_unions
    Feature record: CXX_FEATURE:0cxx_user_literals
    Feature record: CXX_FEATURE:0cxx_variable_templates
    Feature record: CXX_FEATURE:0cxx_variadic_macros
    Feature record: CXX_FEATURE:0cxx_variadic_templates

ChristophHaag, 2021-04-27:
> The source directory

> /home/bozdoc84/SteamVR-OpenHMD/subprojects/openhmd

> does not contain a CMakeLists.txt file.

git submodule update --init

Bozdoc84, 2021-04-27:
all great . thaks ;)