ChristophHaag/SteamVR-OpenHMD #65: errors with "needed by /usr/lib/libopencv_viz.so.4.5.2"

Gladox114, 2021-05-28:
So I am not sure how to compile it exactly for Manjaro Gnome with an another [version of OpenHMD ](https://github.com/thaytan/OpenHMD/tree/rift-kalman-filter) so all I did was replacing that branch with the current submodule but I get errors when compiling.

I did the exact same thing and on the command make it gives this error. https://hastebin.com/zuturokiza.sql
I looked if "/usr/lib/libopencv_viz.so.4.5.2" existed and yeah it exists

Gladox114, 2021-05-28:
compiling everything with meson gives me another error that prevents me from compiling it.

`ninja: Entering directory `build'
[25/63] Compiling C object subprojects/openhmd/libopenhmd.a.p/src_ext_deps_nxjson.c.o
../subprojects/openhmd/src/ext_deps/nxjson.c: In function ‘parse_value’:
../subprojects/openhmd/src/ext_deps/nxjson.c:252:13: warning: ‘new_key’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  252 |           p=parse_value(js, new_key, p, encoder);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[42/63] Compiling C object subprojects/openhmd/libopenhmd.a.p/src_drv_wmr_wmr.c.o
../subprojects/openhmd/src/drv_wmr/wmr.c: In function ‘open_device’:
../subprojects/openhmd/src/drv_wmr/wmr.c:347:20: warning: ‘resolution_v’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  347 |  int resolution_h, resolution_v;
      |                    ^~~~~~~~~~~~
../subprojects/openhmd/src/drv_wmr/wmr.c:347:6: warning: ‘resolution_h’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  347 |  int resolution_h, resolution_v;
      |      ^~~~~~~~~~~~
[55/63] Compiling C++ object driver_openhmd.so.0.0.1.p/driverlog.cpp.o
In file included from ../driverlog.h:9,
                 from ../driverlog.cpp:3:
../subprojects/openvr/openvr_driver.h:2126:8: warning: ‘class vr::IVRSettings’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2126 |  class IVRSettings
      |        ^~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2590:7: warning: ‘class vr::ITrackedDeviceServerDriver’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2590 | class ITrackedDeviceServerDriver
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2640:8: warning: ‘class vr::IVRDisplayComponent’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2640 |  class IVRDisplayComponent
      |        ^~~~~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2687:8: warning: ‘class vr::IVRDriverDirectModeComponent’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2687 |  class IVRDriverDirectModeComponent
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2751:8: warning: ‘class vr::ICameraVideoSinkCallback’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2751 |  class ICameraVideoSinkCallback
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2760:8: warning: ‘class vr::IVRCameraComponent’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2760 |  class IVRCameraComponent
      |        ^~~~~~~~~~~~~~~~~~
In file included from ../driverlog.h:9,
                 from ../driverlog.cpp:3:
../subprojects/openvr/openvr_driver.h:2803:7: warning: ‘class vr::IVRDriverContext’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2803 | class IVRDriverContext
      |       ^~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2816:7: warning: ‘class vr::IServerTrackedDeviceProvider’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2816 | class IServerTrackedDeviceProvider
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2862:7: warning: ‘class vr::IVRWatchdogProvider’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2862 | class IVRWatchdogProvider
      |       ^~~~~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:2878:7: warning: ‘class vr::IVRCompositorPluginProvider’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2878 | class IVRCompositorPluginProvider
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../driverlog.h:9,
                 from ../driverlog.cpp:3:
../subprojects/openvr/openvr_driver.h:2939:7: warning: ‘class vr::IVRProperties’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 2939 | class IVRProperties
      |       ^~~~~~~~~~~~~
In file included from ../driverlog.h:9,
                 from ../driverlog.cpp:3:
../subprojects/openvr/openvr_driver.h:3364:8: warning: ‘class vr::IVRDriverInput’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 3364 |  class IVRDriverInput
      |        ^~~~~~~~~~~~~~
In file included from ../driverlog.h:9,
                 from ../driverlog.cpp:3:
../subprojects/openvr/openvr_driver.h:3399:7: warning: ‘class vr::IVRDriverLog’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 3399 | class IVRDriverLog
      |       ^~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:3422:7: warning: ‘class vr::IVRServerDriverHost’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 3422 | class IVRServerDriverHost
      |       ^~~~~~~~~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:3472:7: warning: ‘class vr::IVRCompositorDriverHost’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
[... 88 lines cut ...]
 3594 | class IVRResources
      |       ^~~~~~~~~~~~
../subprojects/openvr/openvr_driver.h:3644:8: warning: ‘class vr::IVRIOBuffer’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 3644 |  class IVRIOBuffer
      |        ^~~~~~~~~~~
[... cut ...]
      |       ^~~~~~~~~~~~~~~~~~~~
../driver_openhmd.cpp:447:7: warning: base class ‘class vr::IVRDisplayComponent’ has accessible non-virtual destructor [-Wnon-virtual-dtor]
../driver_openhmd.cpp:996:7: warning: base class ‘class vr::IServerTrackedDeviceProvider’ has accessible non-virtual destructor [-Wnon-virtual-dtor]
  996 | class CServerDriver_OpenHMD: public IServerTrackedDeviceProvider
      |       ^~~~~~~~~~~~~~~~~~~~~
[63/63] Linking target subprojects/openhmd/openhmd_simple_example`

thaytan, 2021-05-28:
There are no errors in your meson build that I can see. The build completed successfully.

Gladox114, 2021-05-29:
> There are no errors in your meson build that I can see. The build completed successfully.

Well I thought these are errors because steam can't find my device.  I did this now 3 times again and 1 out of 3 steam said that some add-ons crashed. The rest steamVR just says that I need to plugin my VR headset.

The AUR "steamvr-openhmd-git" worked but it has no tracking so this means the AUR does something I didn't do maybe