ChristophHaag/SteamVR-OpenHMD #66: CV1 sensor not working?

Gregory-Hepicloud, 2021-07-29:
Hi we have install and run steamvr headset working remote move but if move in room not follow and my sensor on wall not start

ps : if you need log or other who can found it?

thx

thaytan, 2021-07-31:
The easiest option is to build the controller-haptics-wip branch of SteamVR-OpenHMD from my repo: https://github.com/thaytan/SteamVR-OpenHMD/tree/controller-haptics-wip

That will pull in the right OpenHMD branch. You also need to make sure you have USB rules to allow access to the sensor devices. https://github.com/OpenHMD/OpenHMD/wiki/Udev-rules-list has some info. The xr-hardware rules are the most complete.

thaytan, 2021-08-06:
If you have any further trouble, let us know or jump on the #openhmd IRC channel on the Libera.chat IRC server

closed 2021-08-06 by thaytan