ChristophHaag/SteamVR-OpenHMD #72: 422 Error in SteamVR

lixiangwuxian, 2022-01-11:
I have a VR headset, and its usb data format is compatible with Oculus Rift DK1, but the display is connected by DP, shown as an  independent monitor. I managed to build this project on Windows, and I modify the openhmd part to let it identify the headset as DK1. It works well when I run the openhmd simple test, but always gets 422 error when i run steamvr with this driver. Is there anything in source code i can edit to prevent this happening?

System: Windows 10 21H1
SteamVR System Report:[https://mega.nz/file/1vQjzIZJ#LMAiT4kA9d8z734xyVpf7anoOq2yO_upYoYj8YURiew](mega)
Build environment: msvc
Modified code: [https://mega.nz/file/1vwF2SiB#ogAeuwOcXhLxgVo-mvmvi66O7y-MAKjHE8mJ3gUMJUA](mega)

thaytan, 2022-01-13:
I don't have any experience with SteamVR-OpenHMD on Windows.

The big difference between SteamVR and the simple example is that SteamVR will try and find and use the display connection, where the simple example just reads telemetry and only communicates with the USB interfaces. I suspect the display link is the most likely reason for SteamVR to get upset, but I can't guess what it doesn't like there.

ChristophHaag, 2022-01-13:
btw you can drag&drop files directly into github text entry boxes: [SteamVR-2022-01-11-AM_10_57_48.txt](https://github.com/ChristophHaag/SteamVR-OpenHMD/files/7866504/SteamVR-2022-01-11-AM_10_57_48.txt)
 
Basically the SteamVR error codes tell you very little, you have to look at the log files. From your `vrcompositor.txt`
```
Tue Jan 11 2022 10:55:26.284 - Reading EDID vendor Id timed out.
Tue Jan 11 2022 10:55:26.284 - CGraphicsDevice Init...
Tue Jan 11 2022 10:55:26.284 - Headset display is on desktop
Tue Jan 11 2022 10:55:26.290 - Invalid adapter output specified! -1 0
Tue Jan 11 2022 10:55:26.290 - Failed to init graphics device
Tue Jan 11 2022 10:55:26.293 - Failed to initialize compositor
Tue Jan 11 2022 10:55:26.293 - Failed to start compositor: VRInitError_Compositor_InvalidOutputDesktop
```
Looks like the compositor can't start up at all.

There is very little info about VRInitError_Compositor_InvalidOutputDesktop on the internet. This looks similar at least https://steamcommunity.com/app/992490/discussions/0/3166519278497611414/#c3166519278498219400 but we don't really have a different mode.

On Linux SteamVR is *very* happy to do direct mode on whatever display very roughly fits some criteria like the resolution specified in the driver.

On Windows it is very possible that more needs to be done to enable direct mode or to tell it not to use direct mode

Looking through https://github.com/ValveSoftware/openvr/blob/master/headers/openvr_driver.h some obvious settings ones to look into

	Prop_EdidVendorID_Int32						= 2011,
	Prop_EdidProductID_Int32					= 2015,

	//-----------------------------------------------------------------------------
	// direct mode keys
	static const char * const k_pch_DirectMode_Section = "direct_mode";
	static const char * const k_pch_DirectMode_Enable_Bool = "enable";
	static const char * const k_pch_DirectMode_Count_Int32 = "count";
	static const char * const k_pch_DirectMode_EdidVid_Int32 = "edidVid";
	static const char * const k_pch_DirectMode_EdidPid_Int32 = "edidPid";

lixiangwuxian, 2022-01-16:
Thanks a lot. I guess this is beyond my ability yet. I would look into in this way

lixiangwuxian, 2022-01-31:
It's done now. I edited Relativty (https://github.com/relativty/Relativty) driver to combine with OpenHMD and it works well. Maybe we can edit this driver refering to the relativty driver (it use extend mode).

closed 2022-01-31 by lixiangwuxian

lixiangwuxian, 2022-01-31:
> Do you have a link to what you did in the Relativty driver. Otherwise, it's hard to know what to do with that information.

I only edited the rotation sensor part in (https://github.com/relativty/Relativty/blob/master/Relativty_Driver/source/Relativty_HMDDriver.cpp), but I think the main point is the difference between the display components of these drivers, and I did not edit the [display component part](https://github.com/relativty/Relativty/blob/master/Relativty_Driver/include/Relativty_components.h) in Relativty.