ChristophHaag/SteamVR-OpenHMD #73: Display is Rotated 90 degrees (Ubuntu) (Samsung Odyssey+)

prototurd, 2022-02-17:
Samsung Odyssey+ HMD. Am able to get into Steam VR and load the home application with one major problem. The display is rotated 90 degrees. Not 90 degrees yaw or 90 degrees pitch, the display is rolled 90 degrees to the side. So far in all of my googling I've found 0 other instances of this happening in anybody else's experience.

Ubuntu 20.04

tfolbrecht, 2022-02-17:
Does setting it to Portrait work?

`gnome-control-center display`

GUI:
Settings -> Display

prototurd, 2022-02-17:
> Does setting it to Portrait work?
> 
> `gnome-control-center display`
> 
> GUI: Settings -> Display

It did not. However, before compiling I replaced the contents of ~/SteamVR-OpenHMD/subprojects/openhmd/src/drv_wmr/wmr.c with [this](https://github.com/OpenHMD/OpenHMD/pull/252) unapproved commit on the OpenHMD project and the issue was resolved.