ChristophHaag/SteamVR-OpenHMD #76: ./register.sh overwrites/deletes backup

AnotherNeko, 2022-06-17:
[running ./register.sh](https://github.com/ChristophHaag/SteamVR-OpenHMD#run-easy-for-linux) more than once seems to delete the backup it had created previously, which goes against the idea of backups, where you probably want to keep all versions but with different names.

ChristophHaag, 2022-06-18:
True, I wrote it as a minimal one off script. I think the quickest way would be to print an error and refuse to do anything if the target file already exists.

AnotherNeko, 2022-06-18:
what I found easier was including (part of) the Unix time in the filename, that way no two files would have the same name. I'll see if I can come up with a decent-looking PR.

AnotherNeko, 2022-06-18:
I tested that change #77 on my system and it seems to work, and it should work on any other system that has the same mv command, although I didn't test it there.