ChristophHaag/SteamVR-OpenHMD #78: SteamVR Driver Download?

PurpleVoidEpic, 2022-07-19:
Where do I download the steamvr driver I have a VRTEK WVR1 id like to use

also how do I setup this I need detailed instructions 

I'm having issues compiling the steamvr driver

AnotherNeko, 2022-07-19:
I'd happily walk you through this step-by-step. Where does your first error appear?

PurpleVoidEpic, 2022-07-19:
CMake Error at C:/Program Files/CMake/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find HIDAPI (missing: HIDAPI_LIBRARY HIDAPI_INCLUDE_DIR)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  subprojects/openhmd/cmake/FindHIDAPI.cmake:35 (find_package_handle_standard_args)
  subprojects/openhmd/CMakeLists.txt:62 (find_package) 

this is the error I get in cmake for windows

AnotherNeko, 2022-07-19:
Your cmake can't access hidapi. Make sure it's installed, then I have no idea what to do next.

PurpleVoidEpic, 2022-07-19:
nvm i fixed it by upgrading to ubuntu im not longer using windows for this