ChristophHaag/SteamVR-OpenHMD #80: Help Compile Driver in .dll and not in ".so" linux (MOD of 90ºFOV @ 135ºFOV - AnonymousVR)

AnonymousVR-VR, 2022-11-02:
Hi, I have been compiling the OpenHMD drivers, I have used SteamVR-OpenHMD for that.

I have finished a MOD in a VRTEK WVR3 viewer, it was 90 degrees FOV and I have converted it to 135 degrees FOV, you can see the MOD here in this community:

https://www.realovirtual.com/foro/topic/51245/mod-visor-vr-los-90o-fov-135o-fov-anonymous-vr

The thing is, I have been able to compile the drivers, all the drivers, but the extension is ".so" for Linux, I need it for Windows in the .DLL extension.

How should I compile to have .dll?

I have to modify the VRTEK WVR3 driver to rotate the screens about 10 degrees, can you help with this as well?

Thanks.

ChristophHaag, 2022-11-03:
I did see the other issue https://github.com/OpenHMD/OpenHMD/issues/346 but haven't replied there...

Did you compile on linux or wsl? Unfortunately I have no idea how to properly compile this for windows. In theory all that should be needed is actually setting up the project with all its dependencies, cmake should automatically create a dll. In practice nobody has described how to do it so far.

If the views are now actually angled or "canted", then SteamVR's default IPD based view matrices probably won't work. (They have some magic for the canted displays of the Index where they give parallel views to the application and use reprojection to correct the perspective but no idea how that actually works in steamvr https://www.valvesoftware.com/en/index/deep-dive/fov).

iirc, openhmd is too limited here and does not have an API to advertise angled displays, nor does it advertise rotated displays properly. In monado we have a bunch of special cases for that and I have some work in progress for it in SteamVR-OpenHMD too https://github.com/ChristophHaag/SteamVR-OpenHMD/pull/58/files#diff-031dcb13147d99d745b4d66db56be84eb87bfc0206cfe23dc4370b7c0eb239e7R619-R658. So with the name string you would recognize the hmd in SteamVR-OpenHMD. Not sure what's the best way to only apply a special case for a modified WVR3 as you probably won't change the usb id...

To actually give SteamVR the angles of the displays would be with SetDisplayEyeToHead like here https://gitlab.freedesktop.org/monado/monado/-/blob/main/src/xrt/state_trackers/steamvr_drv/ovrd_driver.cpp#L1185-1191

Documentation at https://github.com/ValveSoftware/openvr/wiki/vr::ITrackedDeviceServerDriver-Overview Method 2: Call IVRServerDriverHost::SetDisplayEyeToHead

I'm not sure we have a lot of math functions to create such a transformation matrix from angles. Maybe openhmd math has something, otherwise you'd have to add glm or such, or just calculate the matrices somewhere else and hardcode them.

AnonymousVR-VR, 2022-11-03:
Hello ChristophHaag, thanks for all the information offered.

Yes, I have requested assistance in OpenHMD but the information does not arrive, I guess there are not many cases of MODs with tilted screens.

I have to say that the FOV gained is very good and that the technique used gives very good results, you have to use lenses with the clearest possible edges.

The build was on Ubuntu 22.04 using VM VirtualBox virtual machine from Windows 10 pro.

I have not been able to compile from Windows, after adding the hidapi library, the compilation errors are many, I do not know how to solve the problem to be able to do it from Windows and have a .dll as a result, in fact, it is not possible to do anything.

I have to say that I have no programming knowledge, I can only use and do some things, but my knowledge is nothing.

I have to say, that I am trying to find a solution to the driver or screen rotation problem, but I have to ask for help in some way, because I don't think I will be able to write the lines of code to do something like that.

All the help you can give is welcome, I know there are people like you who understand a lot of programming knowledge, if you can help, do what you can in this MOD.

I guess I have to find a way to compile in .DLL, then see how to modify the code to add the functions.

Thank you very much for your attention ChristophHaag.

AnonymousVR-VR, 2022-11-03:
I am trying to understand, does the driver_openhmd.cpp file already have functions to rotate the screens? I am not sure where this file should work.
Is Monado another Api that uses OpenHMD too? It's a bit of a mess for me, forgive my ignorance.

https://github.com/ChristophHaag/SteamVR-OpenHMD/pull/58/files#

Can this file be compiled to get a driver for SteamVR? And how is this file used? It is different from the file included in its place from SteamVR-OpenHMD.

I try to understand a little.

ChristophHaag, 2022-11-14:
> I am trying to understand, does the driver_openhmd.cpp file already have functions to rotate the screens? I am not sure where this file should work.

Not quite, this is a branch that has not been merged to the main branch yet because I wasn't sure if it always does the right thing.

>  Is Monado another Api that uses OpenHMD too? It's a bit of a mess for me, forgive my ignorance.

OpenHMD as a whole is a bunch of "basic" vr headset drivers.

Monado is a full OpenXR runtime. It includes a bunch of its own headset drivers, but it can also make use of OpenHMD's headset drivers.

> Can this file be compiled to get a driver for SteamVR? And how is this file used? It is different from the file included in its place from SteamVR-OpenHMD.

It's a branch that I wanted to show as an example. It only does 90/180° left/right rotations for the displays, because some of the HMDs have their entire display rotated. For example the OSVR HDK2 display is upside down inside the headset.

Angled/"Canted" rotation would need to be added separately.

AnonymousVR-VR, 2022-11-14:
Thanks for the clarifications.

AnonymousVR-VR, 2022-11-24:
Finally I have used the Openvr API, and I have managed with the projectionraw function to create 2 individual functions for each eye and from there configure my screens.

virtual void GetProjectionRaw( EVREye eEye, float *pfLeft, float *pfRight, float *pfTop, float *pfBottom ) 
if (eEye == Eye_Left)  {} else {}}

Thank you very much for your attention.